LibreHardwareMonitor/LibreHardwareMonitor #517: Memory leak (LibreHardwareMonitorLib.dll)

1usmus, 2021-08-19:
**Sensor.cs**
private TimeSpan _valuesTimeWindow = TimeSpan.Zero;
Unfortunately, even this does not solve the problem. 

AMD / Nvidia identical problem.

nangusa, 2021-08-19:
Hi,

I'm using the lib in a service and I can confirm that setting TimeSpan.Zero is working as expected (ie: no increase in memory useage over time) with nVidia cards and the latest 47x drivers. I tested it with RTX 2000 and GTX 1000 series cards. With the previous branch of nVidia drivers yes, you get memory leaks, the problem is at nVidia side and can be reproduced with other monitoring utilities too ( https://www.hwinfo.com/forum/threads/memory-leak-hwinfo-using-10g-memory.6937/)

My previous issue: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/450#issue-898596191

I don't know how it behaves with AMD cards tough, I don't have any at hand to test.

PhyxionNL, 2021-09-28:
Duplicate of #450.

closed 2021-09-28 by PhyxionNL