LibreHardwareMonitor/LibreHardwareMonitor #521: make StorageInfo's Serial field public

Snazzie, 2021-08-23:
In my performance monitoring app, i want to map stored metrics of storage devices via its serial,  but LibreHardwareMonitor only exposes the Drive letter. If i were to map via Drive letter, and user swaps out drive or changes drive letter, the link is broken. Friendly name exists but that isnt unique and in certain senarios we would have a broken link aswell.

Tried getting Serial from drive letter through WMI but havent got any success. Since its already available in WindowsStorage.StorageInfo, could you expose it just so people dont have to find an alternative way to get it?

Unless i missed something and theres already an existing way to access it?

Snazzie, 2021-08-23:
thats assuming serial numbers are unique?!?

PhyxionNL, 2021-08-23:
I have no objection to making more of StorageInfo public, but it's not guaranteed to be a unique number. I once had two Seagate HDDs with the same S/N.

Snazzie, 2021-08-24:
Yeah thats annoying. if its not a problem, i think its still quite useful to expose these info.

No rush on my end, i ended up using reflection to access the private values.

PhyxionNL, 2021-08-24:
> 
> 
> Yeah thats annoying. if its not a problem, i think its still quite useful to expose these info.
> 

Feel free to submit a PR 👍

closed 2022-09-23 by PhyxionNL