LibreHardwareMonitor/LibreHardwareMonitor #524: LibreHardwareMonitorLib 0.8.8 doesn't work with x86

ts-korhonen, 2021-08-28:
I'm using LibreHardwareMonitorLib in a plugin that need to be compiled in x86.
Upgrading from 0.8.7 to 0.8.8 introduces dependency to NvAPIWrapper that doesn't work in x86 on a x64 environment.

Here is the exception thrown:
```
NvAPIWrapper.Native.Exceptions.NVIDIANotSupportedException: 32bit process running in a 64bit environment can't access NVIDIA API.
   at NvAPIWrapper.Native.Helpers.DelegateFactory.NvAPI_QueryInterface(UInt32 interfaceId)
   at NvAPIWrapper.Native.Helpers.DelegateFactory.GetDelegate[T]()
   at NvAPIWrapper.Native.GeneralApi.Initialize()
   at LibreHardwareMonitor.Hardware.Gpu.NvidiaGroup..ctor(ISettings settings)
   at LibreHardwareMonitor.Hardware.Computer.set_IsGpuEnabled(Boolean value)
```

uxsoft, 2021-09-05:
I have the same issue

PhyxionNL, 2021-09-06:
Any reason you can't make 64-bit version? It's seems like a limitation of NvAPIWrapper, there's not much possible there.

ts-korhonen, 2021-09-06:
> Any reason you can't make 64-bit version? It's seems like a limitation of NvAPIWrapper, there's not much possible there.

The program that host's the plugin is 32-bit and there exists no 64-bit executable of it. 32-bit executables cannot load 64-bit dlls.

xingchuanzhen, 2021-09-07:
I also have this problem

namazso, 2021-09-07:
it seems more like an nvapi limitation: https://github.com/falahati/NvAPIWrapper/blob/master/NvAPIWrapper/Native/Helpers/DelegateFactory.cs#L54-L65
probably not fixable

PhyxionNL, 2021-09-28:
Please give the above PR a try and see if it's working again. I've restored as much as possible from LHM's original implementation and added the remaining bits to keep the features on par with NvAPIWrapper. I don't have multiple GPUs here to try it on, but on my one card I see the same number of sensors now with this version as with the NvAPIWrapper (actually 1 less as NvAPIWrapper incorrectly added a sensor even though NV API reported the sensor as not present)

ts-korhonen, 2021-09-28:
The PR does make it work in the sense that it doesn't throw exception anymore, but it's also missing the Nvidia GPU in x86.

PhyxionNL, 2021-09-28:
> 
> 
> The PR does make it work in the sense that it doesn't throw exception anymore, but it's also missing the Nvidia GPU in x86.

Please try the latest build, I've tried under 32-bit mode here and it works for me.

ts-korhonen, 2021-09-28:
With the latest changes the GPU is back and thus the problem is solved.

PR #553 (2021-09-29): NV API changes
Restores parts of https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/commit/eea8558825cc0d5a76aa3e57fcc862b5d6f22100 while implementing the missing bits from the original implementation.

Fixes #524.
commit 8c01e4b2 (2021-09-29): NV API changes (#553)

* NV API changes

Restores parts of https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/commit/eea8558825cc0d5a76aa3e57fcc862b5d6f22100 while implementing the missing bits from the original implementation.

Fixes #524.
closed 2021-09-29 by PhyxionNL