LibreHardwareMonitor/LibreHardwareMonitor #533: ASUS Crosshair sensors

zeule, 2021-09-08:
I was contacted by @kedema, who runs Crosshair VIII Formula, with a few questions related to finding sensor registers for ASUS motherboards, and we discovered a few things about the Nuvoton chips in these boards as well as mistakes in the current implementation  for them. The boards use the NCT6798D chip, which has no freely available datasheet as of now. Additionally, they have and embedded controller that provides sensor measurements. I don't know the chip name, is it a separate chip, or is it a part of the Nuvoton, or the TPU chip.

I'm unable to find high-resolution photos of the board with covers taken off.  Maybe when I disassemble my system for cleaning and refilling in a few weeks I will take such photos myself. 

We have discovered that 0xC33 and 0xC35 registers of the Nuvoton chip, that are labeled as "Water In" and "Water Out" in LHM now, are actually programmable registers and report temperature for  the Q-Fan sources, namely:
0xC2F:  Chassis 1
0xC33: Chassis 2 
0xC35: Chassis 3 
0xC37:  High Amp 
0xC39 :W+_Pump

While it makes sense to publish these registers as LHM sensors, to read water temperatures one needs to know how to switch the Embedded Controller (EC) banks, as those sensors are reported at 0x00 and 0x01 registers of the bank 1 (0-based index). On the other hand, reading the EC banks directly, via ports 0x62 and 0x66, are known to case problems due to a race condition with other monitoring tools and probably the ASUS AISuite. If there is a way to find these sensors in the Nuvoton chip, that would be nice. However, basing on LHM reports, which contain snapshots of the all Nuvoton registers, there is no "Water In" sensor there (at least I have found no number equal to the water temperature in the report).

zeule, 2021-09-09:
DSDT for C8H contains the following interesting fragment:
```
Name (ECBK, Zero)
Method (BREC, 1, Serialized)
{
    CreateByteField (Arg0, Zero, WLEN)
    B_CT = (WLEN >> 0x02)
    If ((B_CT > 0x20))
    {
        Return (Ones)
    }

    If ((Acquire (ASMX, 0xFFFF) == Zero))
    {
        IDBF = STOH (Arg0)
        Local0 = Zero
        Local1 = Zero
        While ((Local0 < B_CT))
        {
            B_BK = DerefOf (IDBF [Local0])
            Local0++
            B_ID = DerefOf (IDBF [Local0])
            Local0++
            ECBK = \_SB.PCI0.SBRG.EC0.EBFF
            \_SB.PCI0.SBRG.EC0.EBFF = B_BK /* \AMW0.B_BK */
            ODBF [Local1] = \_SB.PCI0.SBRG.EC0.ECCM (B_ID, Zero, Zero)
            \_SB.PCI0.SBRG.EC0.EBFF = ECBK /* \AMW0.ECBK */
            Local1++
        }

        OSBF = HTOS (ODBF, Local1)
        Release (ASMX)
        Return (OSBF) /* \AMW0.OSBF */
    }

    Return (Ones)
}

Method (BWEC, 1, Serialized)
{
    CreateByteField (Arg0, Zero, WLEN)
    B_CT = (WLEN >> 0x02)
    If ((B_CT > 0x20))
    {
        Return (Ones)
    }

    If ((Acquire (ASMX, 0xFFFF) == Zero))
    {
        IDBF = STOH (Arg0)
        Local0 = Zero
        While ((Local0 < B_CT))
        {
            B_BK = DerefOf (IDBF [Local0])
            Local0++
            B_ID = DerefOf (IDBF [Local0])
            Local0++
            B_DT = DerefOf (IDBF [Local0])
            Local0++
            ECBK = \_SB.PCI0.SBRG.EC0.EBFF
            \_SB.PCI0.SBRG.EC0.EBFF = B_BK /* \AMW0.B_BK */
            \_SB.PCI0.SBRG.EC0.ECCM (B_ID, One, B_DT)
            \_SB.PCI0.SBRG.EC0.EBFF = ECBK /* \AMW0.ECBK */
        }

        Release (ASMX)
        Return (Zero)
    }

    Return (Ones)
}
```

Both methods access the embedded controller (device `EC0`).  My guess is `BREC` and `BWEC` stand for "block read EC" and "block write EC". There are corresponding methods for the Super I/O chip (BRIO and BWIO). If these functions are available for other ASUS boards, we should switch to use them rather then the direct ISA ports for Super I/O chips and the EC, but in any case the following fragment seems to perform reading from bank `B_BK` and register `B_ID` backing up current bank number to `ECBK` and restoring it afterwards:
```
ECBK = \_SB.PCI0.SBRG.EC0.EBFF 
\_SB.PCI0.SBRG.EC0.EBFF = B_BK /* \AMW0.B_BK */
ODBF [Local1] = \_SB.PCI0.SBRG.EC0.ECCM (B_ID, Zero, Zero)
\_SB.PCI0.SBRG.EC0.EBFF = ECBK /* \AMW0.ECBK */
```

Which means to switch EC bank number, the bank number has to be written into register 0xFF.

On this stage we probably need DSDT dumps from more ASUS motherboards.

zeule, 2021-09-09:
@kedema, @ReggX , @Pomianowski, @berniyh, @holgerpieta, could you share DSDTs for your ASUS boards, please?

zeule, 2021-09-09:
Found acpidumps in the Linux kernel bug 204807 for these boards: 'Prime X570-P',  'ROG CROSSHAIR VIII IMPACT',  'TUF B550M',  'TUF GAMING X570 PLUS (Wi-Fi)' , 'TUF Z390M-PRO GAMING',  'X570 TUF Gaming Pro',  'X570-Plus TUF Gaming'

malcolmmp, 2021-09-09:
Hi there. I just came across this project recently and really appreciate the work you guys are doing on it.

I've got a Crosshair 8 Impact board and I was in the process of adding it in a pull request to try and make it as easy as possible for the core team, but I believe I may have come across a related issue. I cloned the latest code and compiled with my own basic changes (pretty much consisting of adding the board name in the appropriate places) and this got Flow and Water In working. Water Out isn't reading properly, though it looks like the value may be being reported on Aux3.

Judging from the current active discussion, it sounded related.

I'd like to offer my time and hardware to test anything being worked on. I'm not a C# guy, but if there are any tools you can point me at for prodding around the NCT6798D or dumping any data from it, I'd be happy to do whatever I can to help.

kedema, 2021-09-09:
@zeule There is the DSDT from C8F, let me know if you need anything else.
[DSDT - C8F - Asus crosshair formula.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/7138710/DSDT.-.C8F.-.Asus.crosshair.formula.zip)

I have found addresses that report the same values as previously found, for information:
0xC2F = 0x073: Chassis 1
0xC33: Chassis 2
0xC35 = 0x079: Chassis 3
0xC37 = 0x07b: High Amp
0xC39 :W+_Pump

But these addresses are supposed to be PECI and AUXx temperatures with decimals if I understand correctly, right? 
Example: SourceNct67Xxd.AUXTIN0, 0x079, 0x07A,... 
The CPUTIN, SYSTIN (M.B.) and PECI_0_CAL sensors report good values for me but not for the AUX, how is it for you?

@malcolmmp Yes we have the same problem I think, I stumbled upon it while trying to do the same thing as you for Crosshair 8 Formula

ReggX, 2021-09-09:
@zeule 
[ASUS Crosshair VIII Dark Hero DSDT.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/7139141/ASUS.Crosshair.VIII.Dark.Hero.DSDT.zip)

zeule, 2021-09-09:
Thank you, guys, for the uploaded DSDT files!

All x570 boards seem to provide the same WMI methods for reading Super I/O and EC data. B550 ones provide a dummy functions for the EC. Those block read function use some encoding on the input to pass array of addresses to read from and the array length in a string. I did not look into that yet, but there should not be anything complicated. So, I basically know now how to read sensors data via the WMI functions, but in Linux, where kernel functions to execute WMI methods are simple and there are numerous examples.

But with Windows I still don't understand how to proceed. Looks like one needs to produce a MOF file with interface definitions and  publish it to use with the wmiacpi.sys , or use a custom kernel driver (maybe extend the one already used by LHM). Alternatively, ASUS publishes some hardware monitor methods for the WMI system, and they are available under `root/WMI/ASUSHW`. However, at least in my case, the naming for those functions is meaningless. You can inspect them with the following command in powershell : `Get-CimClass -namespace root/WMI -ClassName ASUSHW | %{$_.CimClassMethods}`. For C8H there are functions named `asio_hw_funXX` and I'm not sure what WMI methods they map to. It seems like `asio_hw_fun21` reads from the Super I/O chip register (Nuvoton), and perhaps  `asio_hw_fun23` is mapped to `BREC` (reads from the EC).

Also, I confirm that writing bank number to the register 0xFF indeed switches EC banks and I was able to read "Water In" and "Water Out" temperatures from register 0x0 and 0x1 in bank 1. Values match those showed by AISuite.

zeule, 2021-09-09:
For the reference and for those who want to try inspect ASUS WMI using PowerShell: https://maidavale.org/blog/investigating-asus-wmi-sensors-from-powershell/

holgerpieta, 2021-09-10:
I tried the ACPICA binary tools: https://acpica.org/downloads/binary-tools
But got nowhere useful. If it is still useful and someone would give me slightly more detailed instructions, I'd be happy to provide DSDT tables for my ASUS TUF Gaming B550M Plus (WiFi).
I also tried the WMI sensors instructions from the post above, but for my board the ASUSHW class seems to have not a single sensor related function, only smbus and asiohw.

ReggX, 2021-09-10:
>You can inspect them with the following command in powershell : `Get-CimClass -namespace root/WMI -ClassName ASUSHW | %{$_.CimClassMethods}`. 

Ouput from my Crosshair VIII Dark Hero system:

```
PS C:\Windows\system32> Get-CimClass -namespace root/WMI -ClassName ASUSHW | %{$_.CimClassMethods}

Name              ReturnType Parameters                   Qualifiers
----              ---------- ----------                   ----------
read_smbus_byte      Boolean {cmd, slave, data, Result}   {Implemented, WmiMethodId}
write_smbus_byte     Boolean {cmd, data, slave, Result}   {Implemented, WmiMethodId}
read_smbus_word      Boolean {cmd, slave, data, Result}   {Implemented, WmiMethodId}
write_smbus_word     Boolean {cmd, data, slave, Result}   {Implemented, WmiMethodId}
read_smbus_block     Boolean {cmd, slave, data, Result}   {Implemented, WmiMethodId}
write_smbus_block    Boolean {cmd, count, data, slave...} {Implemented, WmiMethodId}
asio_hw_fun07        Boolean {wPort, bData}               {Implemented, WmiMethodId}
asio_hw_fun08        Boolean {bData, wPort}               {Implemented, WmiMethodId}
asio_hw_fun42        Boolean {wPort, wData}               {Implemented, WmiMethodId}
asio_hw_fun43        Boolean {wData, wPort}               {Implemented, WmiMethodId}
asio_hw_fun44        Boolean {wPort, dwData}              {Implemented, WmiMethodId}
asio_hw_fun45        Boolean {dwData, wPort}              {Implemented, WmiMethodId}
asio_hw_fun09        Boolean {dwAddr, Width, dwData}      {Implemented, WmiMethodId}
asio_hw_fun10        Boolean {dwAddr, dwData, Width}      {Implemented, WmiMethodId}
asio_hw_fun11        Boolean {Bus, Dev, Func, Reg...}     {Implemented, WmiMethodId}
asio_hw_fun12        Boolean {Bus, Dev, dwData, Func...}  {Implemented, WmiMethodId}
asio_hw_fun13        Boolean {Index, wIndexPort, Data}    {Implemented, WmiMethodId}
asio_hw_fun14        Boolean {Data, Index, wIndexPort}    {Implemented, WmiMethodId}
asio_hw_fun15        Boolean {Index, DataHi, DataLo}      {Implemented, WmiMethodId}
asio_hw_fun16        Boolean {DataHi, DataLo, Index}      {Implemented, WmiMethodId}
asio_hw_fun17        Boolean {Index, Eax, Ebx, Ecx...}    {Implemented, WmiMethodId}
asio_hw_fun28        Boolean {Index, Data}                {Implemented, WmiMethodId}
asio_hw_fun29        Boolean {Data, Index}                {Implemented, WmiMethodId}
asio_hw_fun19        Boolean {Index, LDN, Data}           {Implemented, WmiMethodId}
asio_hw_fun20        Boolean {Data, Index, LDN}           {Implemented, WmiMethodId}
asio_hw_fun21        Boolean {Bank, Index, Data}          {Implemented, WmiMethodId}
asio_hw_fun22        Boolean {Bank, Data, Index}          {Implemented, WmiMethodId}
asio_hw_fun46        Boolean {LdnIndexArray, DataArray}   {Implemented, WmiMethodId}
asio_hw_fun47        Boolean {LdnIndexDataArray}          {Implemented, WmiMethodId}
asio_hw_fun23        Boolean {BankIndexArray, DataArray}  {Implemented, WmiMethodId}
asio_hw_fun24        Boolean {BankIndexDataArray}         {Implemented, WmiMethodId}
asio_hw_fun25        Boolean {BankIndexArray, DataArray}  {Implemented, WmiMethodId}
asio_hw_fun26        Boolean {BankIndexDataArray}         {Implemented, WmiMethodId}

```

Do names match between different boards/chipsets?

zeule, 2021-09-10:
> If it is still useful and someone would give me slightly more detailed instructions, I'd be happy to provide DSDT tables for my ASUS TUF Gaming B550M Plus (WiFi).

Yes, please:
```
.\acpidum.exe -b
.\iasl.exe dsdt.dat
```
This should give you dsdt.dsl file, which I'm asking for.
> I also tried the WMI sensors instructions from the post above, but for my board the ASUSHW class seems to have not a single sensor related function, only smbus and asiohw.

Seems like ASUS removed high-level functions from the BIOS in the motherboards based on the 5-th series AMD chipsets. I'm trying to learn how to use asio_hw functions to read the data.

zeule, 2021-09-10:
> Do names match between different boards/chipsets?

Output from Crosshair VIII Hero:
```
PS C:\Windows\system32> Get-CimClass -namespace root/WMI -ClassName ASUSHW | %{$_.CimClassMethods}

Name              ReturnType Parameters                   Qualifiers
----              ---------- ----------                   ----------
read_smbus_byte      Boolean {cmd, slave, data, Result}   {Implemented, WmiMethodId}
write_smbus_byte     Boolean {cmd, data, slave, Result}   {Implemented, WmiMethodId}
read_smbus_word      Boolean {cmd, slave, data, Result}   {Implemented, WmiMethodId}
write_smbus_word     Boolean {cmd, data, slave, Result}   {Implemented, WmiMethodId}
read_smbus_block     Boolean {cmd, slave, data, Result}   {Implemented, WmiMethodId}
write_smbus_block    Boolean {cmd, count, data, slave...} {Implemented, WmiMethodId}
asio_hw_fun07        Boolean {wPort, bData}               {Implemented, WmiMethodId}
asio_hw_fun08        Boolean {bData, wPort}               {Implemented, WmiMethodId}
asio_hw_fun42        Boolean {wPort, wData}               {Implemented, WmiMethodId}
asio_hw_fun43        Boolean {wData, wPort}               {Implemented, WmiMethodId}
asio_hw_fun44        Boolean {wPort, dwData}              {Implemented, WmiMethodId}
asio_hw_fun45        Boolean {dwData, wPort}              {Implemented, WmiMethodId}
asio_hw_fun09        Boolean {dwAddr, Width, dwData}      {Implemented, WmiMethodId}
asio_hw_fun10        Boolean {dwAddr, dwData, Width}      {Implemented, WmiMethodId}
asio_hw_fun11        Boolean {Bus, Dev, Func, Reg...}     {Implemented, WmiMethodId}
asio_hw_fun12        Boolean {Bus, Dev, dwData, Func...}  {Implemented, WmiMethodId}
asio_hw_fun13        Boolean {Index, wIndexPort, Data}    {Implemented, WmiMethodId}
asio_hw_fun14        Boolean {Data, Index, wIndexPort}    {Implemented, WmiMethodId}
asio_hw_fun15        Boolean {Index, DataHi, DataLo}      {Implemented, WmiMethodId}
asio_hw_fun16        Boolean {DataHi, DataLo, Index}      {Implemented, WmiMethodId}
asio_hw_fun17        Boolean {Index, Eax, Ebx, Ecx...}    {Implemented, WmiMethodId}
asio_hw_fun28        Boolean {Index, Data}                {Implemented, WmiMethodId}
asio_hw_fun29        Boolean {Data, Index}                {Implemented, WmiMethodId}
asio_hw_fun19        Boolean {Index, LDN, Data}           {Implemented, WmiMethodId}
asio_hw_fun20        Boolean {Data, Index, LDN}           {Implemented, WmiMethodId}
asio_hw_fun21        Boolean {Bank, Index, Data}          {Implemented, WmiMethodId}
asio_hw_fun22        Boolean {Bank, Data, Index}          {Implemented, WmiMethodId}
asio_hw_fun46        Boolean {LdnIndexArray, DataArray}   {Implemented, WmiMethodId}
asio_hw_fun47        Boolean {LdnIndexDataArray}          {Implemented, WmiMethodId}
asio_hw_fun23        Boolean {BankIndexArray, DataArray}  {Implemented, WmiMethodId}
asio_hw_fun24        Boolean {BankIndexDataArray}         {Implemented, WmiMethodId}
asio_hw_fun25        Boolean {BankIndexArray, DataArray}  {Implemented, WmiMethodId}
asio_hw_fun26        Boolean {BankIndexDataArray}         {Implemented, WmiMethodId}
```

Function declarations seem to match those for C8DH.

holgerpieta, 2021-09-10:
Here we go:
[dsdt-ASUS_TUF_GAMING_B550M_PLUS_WIFI.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/7145076/dsdt-ASUS_TUF_GAMING_B550M_PLUS_WIFI.zip)

zeule, 2021-09-10:
Here is a MOF source for the ASUSHW object as shown by  WMI Explorer with method ids converted to hex notation:
```

[WMI, dynamic: ToInstance, provider("WmiProv"), guid("{466747a0-70ec-11de-8a39-0800200c9a66}")]
class ASUSHW
{
	[key, read] string InstanceName;
	[read] boolean Active;
	[WmiMethodId(0x424D5352), Implemented] void read_smbus_byte([in] uint32 slave, [in] uint32 cmd, [out] uint32 Result, [out] uint32 data);
	[WmiMethodId(0x424D5357), Implemented] void write_smbus_byte([in] uint32 slave, [in] uint32 cmd, [in] uint32 data, [out] uint32 Result);
	[WmiMethodId(0x574D5352), Implemented] void read_smbus_word([in] uint32 slave, [in] uint32 cmd, [out] uint32 Result, [out] uint32 data);
	[WmiMethodId(0x574D5357), Implemented] void write_smbus_word([in] uint32 slave, [in] uint32 cmd, [in] uint32 data, [out] uint32 Result);
	[WmiMethodId(0x4B4D5352), Implemented] void read_smbus_block([in] uint32 slave, [in] uint32 cmd, [out] uint32 Result, [out, MAX(32)] uint8 data[]);
	[WmiMethodId(0x4B4D5357), Implemented] void write_smbus_block([in] uint32 slave, [in] uint32 cmd, [in] uint32 count, [in, MAX(32)] uint8 data[], [out] uint32 Result);
	[WmiMethodId(0x52494F42), Implemented] void asio_hw_fun07([in] uint16 wPort, [out] uint8 bData);
	[WmiMethodId(0x57494F42), Implemented] void asio_hw_fun08([in] uint16 wPort, [in] uint8 bData);
	[WmiMethodId(0x52494F57), Implemented] void asio_hw_fun42([in] uint16 wPort, [out] uint16 wData);
	[WmiMethodId(0x57494F57), Implemented] void asio_hw_fun43([in] uint16 wPort, [in] uint16 wData);
	[WmiMethodId(0x52494F44), Implemented] void asio_hw_fun44([in] uint16 wPort, [out] uint32 dwData);
	[WmiMethodId(0x57494F44), Implemented] void asio_hw_fun45([in] uint16 wPort, [in] uint32 dwData);
	[WmiMethodId(0x524D454D), Implemented] void asio_hw_fun09([in] uint32 dwAddr, [in] uint8 Width, [out] uint32 dwData);
	[WmiMethodId(0x574D454D), Implemented] void asio_hw_fun10([in] uint32 dwAddr, [in] uint8 Width, [in] uint32 dwData);
	[WmiMethodId(0x52504349), Implemented] void asio_hw_fun11([in] uint8 Bus, [in] uint8 Dev, [in] uint8 Func, [in] uint16 Reg, [in] uint8 Width, [out] uint32 dwData);
	[WmiMethodId(0x57504349), Implemented] void asio_hw_fun12([in] uint8 Bus, [in] uint8 Dev, [in] uint8 Func, [in] uint16 Reg, [in] uint8 Width, [in] uint32 dwData);
	[WmiMethodId(0x52494458), Implemented] void asio_hw_fun13([in] uint16 wIndexPort, [in] uint8 Index, [out] uint8 Data);
	[WmiMethodId(0x57494458), Implemented] void asio_hw_fun14([in] uint16 wIndexPort, [in] uint8 Index, [in] uint8 Data);
	[WmiMethodId(0x52444D52), Implemented] void asio_hw_fun15([in] uint32 Index, [out] uint32 DataHi, [out] uint32 DataLo);
	[WmiMethodId(0x57524D52), Implemented] void asio_hw_fun16([in] uint32 Index, [in] uint32 DataHi, [in] uint32 DataLo);
	[WmiMethodId(0x52434944), Implemented] void asio_hw_fun17([in] uint32 Index, [out] uint32 Eax, [out] uint32 Ebx, [out] uint32 Ecx, [out] uint32 Edx);
	[WmiMethodId(0x52444543), Implemented] void asio_hw_fun28([in] uint8 Index, [out] uint8 Data);
	[WmiMethodId(0x57524543), Implemented] void asio_hw_fun29([in] uint8 Index, [in] uint8 Data);
	[WmiMethodId(0x5253494F), Implemented] void asio_hw_fun19([in] uint8 LDN, [in] uint8 Index, [out] uint8 Data);
	[WmiMethodId(0x5753494F), Implemented] void asio_hw_fun20([in] uint8 LDN, [in] uint8 Index, [in] uint8 Data);
	[WmiMethodId(0x5248574D), Implemented] void asio_hw_fun21([in] uint8 Bank, [in] uint8 Index, [out] uint8 Data);
	[WmiMethodId(0x5748574D), Implemented] void asio_hw_fun22([in] uint8 Bank, [in] uint8 Index, [in] uint8 Data);
	[WmiMethodId(0x4252494F), Implemented] void asio_hw_fun46([in] String LdnIndexArray, [out] String DataArray);
	[WmiMethodId(0x4257494F), Implemented] void asio_hw_fun47([in] String LdnIndexDataArray);
	[WmiMethodId(0x4252484D), Implemented] void asio_hw_fun23([in] String BankIndexArray, [out] String DataArray);
	[WmiMethodId(0x4257484D), Implemented] void asio_hw_fun24([in] String BankIndexDataArray);
	[WmiMethodId(0x42524543), Implemented] void asio_hw_fun25([in] String BankIndexArray, [out] String DataArray);
	[WmiMethodId(0x42574543), Implemented] void asio_hw_fun26([in] String BankIndexDataArray);
};
```

Now  `asio_hw_fun25()` is probably mapped to `BREC` , but when I attempt to read EC using it, the call hangs up and AISuite stops updating too. `asio_hw_fun28()` must be `RDEC` and its outputs confirm that.

malcolmmp, 2021-09-10:
Got a dsdt dump for the Crosshair Impact.

Also, for what it's worth, I'm seeing the same set of CIM/WMI class and method names listed as the other Crosshair boards (read/write smbus plus the same asio_hw_fun numbers). No handy sensor_get functions.

[dsdt_Crosshair_VIII_Impact.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/7146653/dsdt_Crosshair_VIII_Impact.zip)

zeule, 2021-09-11:
Thank you all who provided the DSDT data! I need to double check, but from a quick look it seems all of them for the AMD 500 series chipsets do provide the same WMI functions, and the only significant for us difference is that EC reading functions  in B550 BIOSes do nothing (but exist).

Now, returning to the WMI function and their arguments. Using the "acpiexec" utility I verified that the input string for `BREC` should be encoded as follows.
1. ASCII value of the first character must be equal to the length of the rest of the string (or the length of the meaningful data  inside it) in characters.
2. The following string is a sequence of pairs of 4-character blocks, each 4-characters define bank and index in ASCII hex notation (example for bank 1, index 0x25: "00010025").

Thus, to read, for instance, the "water in" sensor value for C8H (EC bank 1, index 0) I need to pass the following string to `asio_hw_fun25()`: "<here is ASCII symbol 8, backspace>00010000".

The next part of the quest is to find out how the MOF for this ASUSHW object appears in the Windows installation? Does AISuite install it? Is it auto-generated somehow by Windows?

zeule, 2021-09-11:
Nope, that encoding scheme is wrong. They expect UTF-16 strings on input, so:

2. The following string is a sequence of pairs of 4-byte blocks, each 8-byte block defines bank and index in UTF-16 hex notation.
Example: to read EC registers 0x3A to 0x3D from the bank 0, one has to pass the following ASCII string: " 0000030A0000030B0000030C0000030D"

malcolmmp, 2021-09-11:
I know it's been mentioned that there's parallel work being done on the Linux kernel to support reading the same sensors. I was looking around over there and while I'm not sure what the current state of affairs is, I thought the nct6775 hwmon stuff (which includes nct6798 code) looked interesting:

https://github.com/torvalds/linux/blob/master/drivers/hwmon/nct6775.c

Looks like they're constructing UTF-16 strings just like you said, which makes perfect sense. Maybe some of what's in there as well as changes in the pipeline could be of use.

zeule, 2021-09-12:
Thank you, still can't understand how to make that BREC() call on Windows. It seems to work on Linux, I just tested it with the acpi_call module:
```
# want to read EC registers 0x3A, 0x3B, 0x3C, and 0x3D, that is "Chipset", "CPU", "Motherboard", and "T Sensor" values. All are temperatures in degree Celcius
$ echo '\AMW0.BREC " 0000030A0000030B0000030C0000030D0"' > /proc/acpi/call
# now read back result
$ cat /proc/acpi/call                                                      
{0x10, 0x00, 0x33, 0x00, 0x41, 0x00, 0x31, 0x00, 0x45, 0x00, 0x32, 0x00, 0x32, 0x00, 0x31, 0x00, 0x38, 0x00, 0x30, 0x30, 0x30, 0x30, 0x33, 0x30, 0x43, 0x30, 0x30, 0x30, 0x30
, 0x30, 0x33, 0x30, 0x44, 0x30, 0x00, 0x00, 0x00, 0x00, 0x00, 0x00, ...
```

OK, it replied with a buffer containing UTF-16 string of length 0x10 bytes, which is 8 characters, which in turn is 4 numeric values. Seems correct. Here are these 16 bytes: 
0x33, 0x00, 0x41, 0x00, 0x31, 0x00, 0x45, 0x00, 0x32, 0x00, 0x32, 0x00, 0x31, 0x00, 0x38, 0x00. These bytes are UTF-16 codes for "3", "A", "1", "E", "2", "2", "1", "8". Thus the reply contains 4 numbers: 0x3A, 0x1E, 0x22, 0x18, or 58, 30, 34, 24 in decimal notation. 

Chipset 58 ℃
CPU 30 ℃
Motherboard 34 ℃
T Sensor  24 ℃

Probably yes, they seem legit. T Sensor value is certainly correct. 

Repeating calls do not seem to lock or slow down. This seems to work just fine. Now, how to make it work on Windows? Otherwise, my Linux driver providing HWMON sensors for ASUS AM-4 500 series motherboards will be working before LHM learns how to make WMI calls to the ASUS firmware... I've tried to call "asio_hw_fun25" from .NET using `InvokeMethod()`  from `System.Management.ManagementObject`, but when I pass a single string there it replies with invalid parameters exception. Calling from the powershell does not seem to work too: the call does not return (although does not result in an exception), and AISuite updates get delayed for a good minute or two, and the PC does not turn off during that call. It's incredible how API design by Microsoft can complicate things...

zeule, 2021-09-13:
#540 adds EC bank switching and correct water temperature sensors for Hero and I hope that works for Dark Hero as well.

zeule, 2021-09-14:
Implemented reading sensor data from EC as a Linux hwmon driver thar calls the WMI function and it works. Now how the f... to call it correctly in Windows?

zeule, 2021-09-15:
> The next part of the quest is to find out how the MOF for this ASUSHW object appears in the Windows installation? Does AISuite install it? Is it auto-generated somehow by Windows?

It comes from the DSDT, where the `WQMO` variable keeps it in the compiled form. One can extract it from there and use the wmimofck.exe tool to generate a C header file with declarations for those functions and argument types. Although that data itself brings nothing new to our case, it confirms all the information already presented in this thread. But the important bit is that the MOF comes from the BIOS and as such that WMI object (ASUSHW) and its methods are are always present (as opposite to the hypothesis that AISuite installer brings in the MOF definitions).

berniyh, 2021-10-05:
> @berniyh, could you share DSDTs for your ASUS boards, please?

Sorry, a bit late, since I didn't have the opportunity to do this earlier.
Sensors seem to work on my board already. ;)

[dsdt.dsl.gz](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/7289100/dsdt.dsl.gz)
.

berniyh, 2021-10-06:
hm, weird, I used those same commands (minus the typo on the program name) to get the table.
Let me check again this afternoon.

zeule, 2021-10-06:
@ReggX, sorry, I forgot that hardware report contains a snapshot for all EC registers. Could you instead share the report?

berniyh, 2021-10-06:
Does this look better?
[apic.dsl.gz](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/7293795/apic.dsl.gz)

zeule, 2021-10-06:
iasl.exe dsdt.dat

zeule, 2021-10-06:
> acpidump.exe does not output a dsdt.dat, sorry.

Have no idea how that could be possible: the DSDT table contains, among other things, data and code for handling ACPI power events, without these your PC would not be able to change power levels.

zeule, 2021-10-06:
Thanks! I did not have DSDT for this board, but somehow EC sensors configuration is already implemented here, in LHM, and in the Linux HWMON driver...

berniyh, 2021-10-06:
Well yeah, as I said above, it's working already when I checked with Windows. ;)
Doesn't work using Linux so far, though. At least I couldn't find a way yet.

zeule, 2021-10-06:
> Well yeah, as I said above, it's working already when I checked with Windows and LHM. ;) 

Missed that, pardon me.

> Doesn't work using Linux hwmon so far, though. At least I couldn't find a way yet.

Do you mean [this driver](https://github.com/zeule/asus-wmi-ec-sensors) does not work? If so, could you open an issue there, please?

altsalt, 2022-03-15:
It appears that sensor support for these boards has been added in 5.16

[source](https://www.phoronix.com/scan.php?page=news_item&px=Linux-5.16-HWMON)