LibreHardwareMonitor/LibreHardwareMonitor #536: Set GPU fan speed from lib

psidex, 2021-09-10:
I know how to read the % fan speed from my GPU using `LibreHardwareMonitorLib`, but is there a way to set the fan speed like you can do from `LibreHardwareMonitor` itself?

javguit, 2021-09-11:
IIRC you have to identify which sensors are the fan control sensors ( If (sensor.SensorType == SensorType.Control) ), and then use the method Sensor.SetSoftware(value) to gain control and set the new value, or if you prefer you can do sensor.ControlMode = ControlMode.Software to gain control and after that set the value, but Sensor.SetSoftware(value) will grant you control anyways. Remember always to use Sensor.SetDefualt() when exiting the program or when you don't want to keep manually controlling the fan speed.
The methods can be found in /hardware/control.cs

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/0fd00e27bb806f7ca29081125e5e58371bda841e/LibreHardwareMonitorLib/Hardware/Control.cs#L87-L96

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/0fd00e27bb806f7ca29081125e5e58371bda841e/LibreHardwareMonitorLib/Hardware/Control.cs#L51-L63

PhyxionNL, 2021-09-11:
Please use discussions for this.

closed 2021-09-11 by PhyxionNL

psidex, 2021-09-12:
@javguit Thanks, that's exactly what I needed.

@PhyxionNL sorry about that, if I have any more questions I'll use a discussion :)