LibreHardwareMonitor/LibreHardwareMonitor #548: Corsair command pro

Mantrain, 2021-09-25:
Corsair commander pro has sensors available, seems like its not added to LHWM.

6 fans, 4 temperature probes and 3/5/12v readings.

Lots of information here:; https://github.com/audiohacked/OpenCorsairLink/issues/70

whatisboom, 2021-12-31:
👍 I would like to be able to use my coolant temperature sensor from the Command Pro in https://github.com/Rem0o/FanControl.Releases

Rem0o, 2022-02-09:
@whatisboom I may have commander pro support. If you are willing to give it a shot, I will post a branch soon.

bin101, 2022-04-09:
@whatisboom could you share your progress? I really would like to have my commander pro sensors/fan controls in "FanControl" so I could start adjusting fan speed with a delta temperature between water and ambient temps.

bin101, 2022-04-09:
@Rem0o  could you share your progress? I really would like to have my commander pro sensors/fan controls in "FanControl" so I could start adjusting fan speed with a delta temperature between water and ambient temps.

Rem0o, 2022-04-09:
@bin101  https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/658