LibreHardwareMonitor/LibreHardwareMonitor #566: Incorrect watts and clock speed readings on Ryzen 7 3700X

WhAtEvErYoUmEaN, 2021-10-01:
[image]

Version 0.8.8-ci20

According to the readings i need my own power plant to sustain my new super computer with up to ∞ Mhz

It's been going on for a while (months) and i can't find a build old enough to test for exactly when this behavior started

PhyxionNL, 2021-10-02:
Can you attach the log? Maybe even debug this?

WhAtEvErYoUmEaN, 2021-10-02:
I'd like to help debug it if you can tell me what exactly you need. Got Visual Studio set up for .NET anyway.

I hope you mean the sensor log, that'd be this one
[LibreHardwareMonitorLog-2021-10-02.csv](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/7271533/LibreHardwareMonitorLog-2021-10-02.csv)

ntoff, 2021-10-16:
For anyone that might want to test this themselves, this commit, specifically line 304: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/commit/1cd403a6fa7f846df6f25fd45f885fcf54b18e51#diff-e84f8a6d92f171e1af3e6afd7459caefed458264740faad8cf47b264c07c3ff9R304

completely breaks the ryzen smu table I included above. I don't know if it still works for smaller tables.

WhAtEvErYoUmEaN, 2021-11-04:
[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/7478105/LibreHardwareMonitor.Report.txt)

Report attached for good measure. Anything else i can do to help?

Ririshi, 2021-11-14:
My 3900X has the same issue. Looks funny but not very functional. I'd be happy to attach logs or run some tests!

qdel, 2021-11-14:
Hi, I have added, using the same method as @ntoff support for 3900x.

Some values does not seems to work however (C states by core mainly seems buggy).

I can create a push request with modifications on RyzenSMU.cs.

```c++
0x00240803,
                new Dictionary<uint, SmuSensorType>
                {
                    //Voltages
                    { 40, new SmuSensorType { Name = "CPU VID (Tel)", Type = SensorType.Voltage, Scale = 1 } },
                    { 41, new SmuSensorType { Name = "Vcore (Tel)", Type = SensorType.Voltage, Scale = 1 } },
                    { 45, new SmuSensorType { Name = "SoC Voltage (Tel)", Type = SensorType.Voltage, Scale = 1 } },
                    { 124, new SmuSensorType { Name = "VDDM", Type = SensorType.Voltage, Scale = 1 } },
                    { 125, new SmuSensorType { Name = "VDDP", Type = SensorType.Voltage, Scale = 1 } },
                    { 126, new SmuSensorType { Name = "VDDG", Type = SensorType.Voltage, Scale = 1 } },
                    { 163, new SmuSensorType { Name = "Core #01 (RyzenSMU)", Type = SensorType.Voltage, Scale = 1 } },
                    { 164, new SmuSensorType { Name = "Core #02 (RyzenSMU)", Type = SensorType.Voltage, Scale = 1 } },
                    { 165, new SmuSensorType { Name = "Core #03 (RyzenSMU)", Type = SensorType.Voltage, Scale = 1 } },
                    { 166, new SmuSensorType { Name = "Core #04 (RyzenSMU)", Type = SensorType.Voltage, Scale = 1 } },
                    { 167, new SmuSensorType { Name = "Core #05 (RyzenSMU)", Type = SensorType.Voltage, Scale = 1 } },
                    { 168, new SmuSensorType { Name = "Core #06 (RyzenSMU)", Type = SensorType.Voltage, Scale = 1 } },
                    { 169, new SmuSensorType { Name = "Core #07 (RyzenSMU)", Type = SensorType.Voltage, Scale = 1 } },
                    { 170, new SmuSensorType { Name = "Core #08 (RyzenSMU)", Type = SensorType.Voltage, Scale = 1 } },
                    { 171, new SmuSensorType { Name = "Core #09 (RyzenSMU)", Type = SensorType.Voltage, Scale = 1 } },
                    { 172, new SmuSensorType { Name = "Core #10 (RyzenSMU)", Type = SensorType.Voltage, Scale = 1 } },
                    { 173, new SmuSensorType { Name = "Core #11 (RyzenSMU)", Type = SensorType.Voltage, Scale = 1 } },
                    { 174, new SmuSensorType { Name = "Core #12 (RyzenSMU)", Type = SensorType.Voltage, Scale = 1 } },
                    //Powers
                    { 0, new SmuSensorType { Name = "PPT Limit", Type = SensorType.Power, Scale = 1 } },
                    { 1, new SmuSensorType { Name = "PPT Value", Type = SensorType.Power, Scale = 1 } },
                    { 26, new SmuSensorType { Name = "VDDIO Mem", Type = SensorType.Power, Scale = 1 } },
                    { 29, new SmuSensorType { Name = "Socket Power", Type = SensorType.Power, Scale = 1 } },
                    { 116, new SmuSensorType { Name = "DDR VDDP Power", Type = SensorType.Power, Scale = 1 } },
                    { 120, new SmuSensorType { Name = "IOD VDDIO Mem", Type = SensorType.Power, Scale = 1 } },
                    { 451, new SmuSensorType { Name = "L3 Cache (Logic)", Type = SensorType.Power, Scale = 1 } },
                    { 455, new SmuSensorType { Name = "L3 Cache (VDDM)", Type = SensorType.Power, Scale = 1 } },
                    { 47, new SmuSensorType { Name = "SoC Power (Tel)", Type = SensorType.Power, Scale = 1 } },
                    { 43, new SmuSensorType { Name = "Total Core Power (Tel)", Type = SensorType.Power, Scale = 1 } },
                    { 147, new SmuSensorType { Name = "Core #1 (RyzenSMU)", Type = SensorType.Power, Scale = 1 } },
                    { 148, new SmuSensorType { Name = "Core #2 (RyzenSMU)", Type = SensorType.Power, Scale = 1 } },
                    { 149, new SmuSensorType { Name = "Core #3 (RyzenSMU)", Type = SensorType.Power, Scale = 1 } },
                    { 150, new SmuSensorType { Name = "Core #4 (RyzenSMU)", Type = SensorType.Power, Scale = 1 } },
                    { 151, new SmuSensorType { Name = "Core #5 (RyzenSMU)", Type = SensorType.Power, Scale = 1 } },
                    { 152, new SmuSensorType { Name = "Core #6 (RyzenSMU)", Type = SensorType.Power, Scale = 1 } },
                    { 153, new SmuSensorType { Name = "Core #7 (RyzenSMU)", Type = SensorType.Power, Scale = 1 } },
                    { 154, new SmuSensorType { Name = "Core #8 (RyzenSMU)", Type = SensorType.Power, Scale = 1 } },
                    { 155, new SmuSensorType { Name = "Core #9 (RyzenSMU)", Type = SensorType.Power, Scale = 1 } },
                    { 156, new SmuSensorType { Name = "Core #10 (RyzenSMU)", Type = SensorType.Power, Scale = 1 } },
                    { 157, new SmuSensorType { Name = "Core #11 (RyzenSMU)", Type = SensorType.Power, Scale = 1 } },
                    { 158, new SmuSensorType { Name = "Core #12 (RyzenSMU)", Type = SensorType.Power, Scale = 1 } },
                    //Currents
                    { 2, new SmuSensorType { Name = "TDC Limit", Type = SensorType.Current, Scale = 1 } },
                    { 3, new SmuSensorType { Name = "TDC Value", Type = SensorType.Current, Scale = 1 } },
                    { 15, new SmuSensorType { Name = "TDC Actual", Type = SensorType.Current, Scale = 1 } },
                    { 8, new SmuSensorType { Name = "EDC Limit", Type = SensorType.Current, Scale = 1 } },
[... cut ...]
                    { 298, new SmuSensorType { Name = "Core #08 C6", Type = SensorType.Load, Scale = 1 } },
                    { 299, new SmuSensorType { Name = "Core #09 C6", Type = SensorType.Load, Scale = 1 } },
                    { 300, new SmuSensorType { Name = "Core #10 C6", Type = SensorType.Load, Scale = 1 } },
                    { 301, new SmuSensorType { Name = "Core #11 C6", Type = SensorType.Load, Scale = 1 } },
                    { 302, new SmuSensorType { Name = "Core #12 C6", Type = SensorType.Load, Scale = 1 } },*/
                }
```

ntoff, 2022-12-02:
An update from my side, I don't think getting these values through a predefined SMU table is right / going to work. I just upgraded from a 5600x to a 5800X3D and they both seem to use the same table `00380905` (according to zen timings) at least for me on my motherboard. 

The code below works on the 5800X3D because it has all 8 cores, but for my 5600X, the 2 middle entries (183 and 184) simply don't produce a value, and 185 - 187 are cores 4, 5, and 6, where as on the 5800X3D, 185 - 187 are 6, 7, and 8. You'd have to do some if then else to figure out whether the cpu is 6 or 8 cores. I assume they have 1 table for 6/8 core, another table for 12/16 core? I'm not a programmer so I really have no clue.

```
{ 180, new SmuSensorType { Name = "Core #1 Voltage", Type = SensorType.Voltage, Scale = 1 } },
{ 181, new SmuSensorType { Name = "Core #2 Voltage", Type = SensorType.Voltage, Scale = 1 } },
{ 182, new SmuSensorType { Name = "Core #3 Voltage", Type = SensorType.Voltage, Scale = 1 } },
{ 183, new SmuSensorType { Name = "Core #4 Voltage", Type = SensorType.Voltage, Scale = 1 } }, //doesn't exist on 6 core cpu (5600x)
{ 184, new SmuSensorType { Name = "Core #5 Voltage", Type = SensorType.Voltage, Scale = 1 } }, //doesn't exist on 6 core cpu (5600x)
{ 185, new SmuSensorType { Name = "Core #6 Voltage", Type = SensorType.Voltage, Scale = 1 } }, //core 4 on 6 core cpu (5600x)
{ 186, new SmuSensorType { Name = "Core #7 Voltage", Type = SensorType.Voltage, Scale = 1 } }, //core 5 on 6 core cpu (5600x)
{ 187, new SmuSensorType { Name = "Core #8 Voltage", Type = SensorType.Voltage, Scale = 1 } }, //core 6 on 6 core cpu (5600x)
```

I don't know how other software (hwinfo, msi afterburner) get the info they do with memory integrity turned on. I guess there's a special signed driver?

@PhyxionNL perhaps you could pin this issue so others who come across this issue can easily find it? Maybe someone will know a way to get the original method to work with memory integrity on.

qdel, 2022-12-02:
> The code below works on the 5800X3D because it has all 8 cores, but for my 5600X, the 2 middle entries (183 and 184) simply don't produce a value, and 185 - 187 are cores 4, 5, and 6, where as on [...]

I do believe this is expected. And that the absence of value if totally normal: zen ccd (core package) is made of 8 cores.
5600x have 1 ccd with 2 disabled core.
5800x have 1 full ccd.
5900x have 2 ccd with 2 disabled core each.
5950x have 2 full ccd.

It allows Amd to refurbish ccd depending of their binning as lesser / higher models.
Seems totally normal to me.

brunoherrera, 2026-03-01:
no fix?