LibreHardwareMonitor/LibreHardwareMonitor #568: Controls/Speeds sometimes stuck on Crosshair VIII Dark Hero (NCT6798D)

ReggX, 2021-10-03:
Companion issue to https://github.com/Rem0o/FanControl.Releases/issues/347

CC @zeule @Rem0o 

Since Fan Control V93 I noticed that some Controls get stuck when running both FanControl and a current build of LibreHardwareMonitor.
I run my Water Pump with a custom curve that has 20% intervals, but sometimes the the Pump Control is stuck in a value in-between or on an old value from some time ago (e.g. Curve is at 40%, but Control is stuck at 60%)
[image]
(I also happens for fans, but those are on a linear curve so it's less noticable (sound less distinct compared to the pump intervals) that they spin at the wrong RPM)

* Disabling curve override and re-enabling does not fix the issue. Rather, the default BIOS curve kicks in and FanControl can't take control anymore.
* Restarting FanControl results in some sensors missing (the NCT6798D ones) and the Pump Control is thereby gone completely, refreshing sensors does not help.
* Quitting LHM and cycling the curve override and restarting LHM does help though and brings the Control back to life.

I have also noticed that it happens in reverse (FanControl at 21% fan speed, LHM stuck on 25%) 

No idea if manual intervention is necessary in the first place or if the race gets resolved on its own after some time passes and I just didn't notice it happening in the background. EDIT: Scratch that. While writing this issue, I could see LHM getting stuck and then unstuck itself after some time passed. The same thing happened in reverse right after, but FanControl didn't unstuck itself even after multiple minutes passed.

My suspicion is that is has something to do with recent changes to how SuperIO/NCT6798D values get read introduced with #540 (maybe race condition on bank access? one instance reading while another one is writing sounds like a prime candidate for a race condition)

zeule, 2021-10-03:
#540 made no change to the SuperIO access method.

Rem0o, 2021-10-05:
I don't expect 2 instances of the LHM backend running at the same time to play along nicely.

zeule, 2021-10-05:
@Rem0o, could you elaborate, please? Access to hardware is guarded by a global mutex in LHM, so I can't understand how LHM instances can conflict with each other.

Rem0o, 2021-10-05:
Quite simply if 2 instances don’t ask the same thing to a control. If one expects default behaviour and one does a SetSoftware, that state is not shared. The first instance still thinks the control is bios controlled. I did not play with multiple instances much as I dont think that’s a common use cass, but that’s one thing I spotted.

zeule, 2021-10-06:
@Rem0o, but that can't explain missing sensors, can it?

Rem0o, 2021-10-06:
@zeule Well if some ressources are locked for some reason by 1 instance while the other try to initialize, it might miss some sensors, I don't know, race conditions can be sneaky.

wayneb64, 2022-01-20:
Most of this discussion has gone right over my head but I think I am seeing something similar. I have LHM 0.8.9 running on two PCs, the first is Crosshair Hero VIII WiFi with 3800X and  GTX 1070 and the secondis a Dark Hero with 5800X and a GTC 3070. The first one has no issues but the second one is REALLY jinky. Any time I touch LHM in any way I get glitchy/laggy response. Just moving the gadget on the desktop is very 'unsmooth'. 
I also run AI Suite III to control my fan speeds. 
Is this really just the Dark Hero that is the common culprit? (I am at latest BIOS level)