LibreHardwareMonitor/LibreHardwareMonitor #574: No CPU Temperature reading on AMD Ryzen 5 2600.

d8ahazard, 2021-10-10:
Hi there!

Trying to implement cross-platform CPU/Memory statistics in my app, and I noticed that my CPU is not reporting any temperature values for (Core (Tctl/ Tdie). Tried on the stable 0.8.8 release, as well as the current prerelease version 0.8.9-pre47.

Here's a hardware report:

`LibreHardwareMonitor Report

--------------------------------------------------------------------------------

Version: 0.8.9.0

--------------------------------------------------------------------------------

Common Language Runtime: 5.0.10
Operating System: Microsoft Windows NT 10.0.22468.0
Process Type: 64-Bit

--------------------------------------------------------------------------------

Ring0

Status: Installing driver "E:\Dev\Glimmr\src\Glimmr\bin\Windows\Glimmr.sys" failed and file exists
First Exception: OpenSCManager returned zero.
Second Exception: OpenSCManager returned zero.

--------------------------------------------------------------------------------

Sensors

|
+- MSI B450 TOMAHAWK (MS-7C02) (/motherboard)
|
+- AMD Ryzen 5 2600 (/amdcpu/0)
|  +- Core (SVI2 TFN) :     1.55     1.55     1.55 (/amdcpu/0/voltage/0)
|  +- SoC (SVI2 TFN) :     1.55     1.55     1.55 (/amdcpu/0/voltage/1)
|  +- Core #1 VID    :     1.55     1.55     1.55 (/amdcpu/0/voltage/2)
|  +- Core #2 VID    :     1.55     1.55     1.55 (/amdcpu/0/voltage/3)
|  +- Core #3 VID    :     1.55     1.55     1.55 (/amdcpu/0/voltage/4)
|  +- Core #4 VID    :     1.55     1.55     1.55 (/amdcpu/0/voltage/5)
|  +- Core #5 VID    :     1.55     1.55     1.55 (/amdcpu/0/voltage/6)
|  +- Core #6 VID    :     1.55     1.55     1.55 (/amdcpu/0/voltage/7)
|  +- Package        :        0        0        0 (/amdcpu/0/power/0)
|  +- Core #1 (SMU)  :        0        0        0 (/amdcpu/0/power/1)
|  +- Core #2 (SMU)  :        0        0        0 (/amdcpu/0/power/2)
|  +- Core #3 (SMU)  :        0        0        0 (/amdcpu/0/power/3)
|  +- Core #4 (SMU)  :        0        0        0 (/amdcpu/0/power/4)
|  +- Core #5 (SMU)  :        0        0        0 (/amdcpu/0/power/5)
|  +- Core #6 (SMU)  :        0        0        0 (/amdcpu/0/power/6)
|  +- Core #1        :      NaN      NaN      NaN (/amdcpu/0/clock/1)
|  +- Core #2        :      NaN      NaN      NaN (/amdcpu/0/clock/2)
|  +- Core #3        :      NaN      NaN      NaN (/amdcpu/0/clock/3)
|  +- Core #4        :      NaN      NaN      NaN (/amdcpu/0/clock/4)
|  +- Core #5        :      NaN      NaN      NaN (/amdcpu/0/clock/5)
|  +- Core #6        :      NaN      NaN      NaN (/amdcpu/0/clock/6)
|  +- Core (Tctl/Tdie) :        0        0        0 (/amdcpu/0/temperature/2)
|  +- CPU Total      :  20.2703  20.2703   28.125 (/amdcpu/0/load/0)
|  +- CPU Core #1    :  13.5135  13.5135       75 (/amdcpu/0/load/1)
|  +- CPU Core #2    :  16.2162        0  16.2162 (/amdcpu/0/load/2)
|  +- CPU Core #3    :  29.7297     12.5  29.7297 (/amdcpu/0/load/3)
|  +- CPU Core #4    :  18.9189     12.5  18.9189 (/amdcpu/0/load/4)
|  +- CPU Core #5    :  32.4324  32.4324     37.5 (/amdcpu/0/load/5)
|  +- CPU Core #6    :  10.8108  10.8108       25 (/amdcpu/0/load/6)
|  +- CPU Core #7    :  32.4324  32.4324       50 (/amdcpu/0/load/7)
|  +- CPU Core #8    :  16.2162  16.2162       50 (/amdcpu/0/load/8)
|  +- CPU Core #9    :  13.5135  13.5135     37.5 (/amdcpu/0/load/9)
|  +- CPU Core #10   :  18.9189        0  18.9189 (/amdcpu/0/load/10)
|  +- CPU Core #11   :   27.027   27.027     37.5 (/amdcpu/0/load/11)
|  +- CPU Core #12   :  13.5135        0  13.5135 (/amdcpu/0/load/12)
|  +- Core #1        :      NaN      NaN      NaN (/amdcpu/0/factor/0)
[... 1102 lines cut ...]
Download Speed
50027.477

Network Utilization
0.07407278

Hyper-V Virtual Ethernet Adapter #2
Up

Data Uploaded
0.25021738

Data Downloaded
1.0153279E-05

Upload Speed
0

Download Speed
0

Network Utilization
0

`

gamelaster, 2021-12-26:
I have same behavior for my Ryzen 3950X ☹️

gamelaster, 2021-12-26:
I found solution, you need to run your app as Administrator. If you will have any other questions, let me know

darkbrain-fc, 2021-12-26:
> ID: <LibreHardwareMonitor/LibreHardwareMonitor/issues/574/1001214081@
> github.com>

Marketing node on: That's why I've built Hardware Supervisor
https://darkbrain-fc.github.io/HardwareSupervisor/ ;-)

ttimasdf, 2023-07-18:
> Marketing node on: That's why I've built Hardware Supervisor https://darkbrain-fc.github.io/HardwareSupervisor/ ;-)

That's EXACTLY what I want! I'm already thinking about writing my own one! Thanks dude!