LibreHardwareMonitor/LibreHardwareMonitor #598: New nvidia GPU monitoring causing stutter

xcasxcursex, 2021-11-11:
To start: This is not a bug unique to librehardwaremonitor. All indications are of an API bug. I am reporting this in the hope that this issue can be pushed upstream from here. Yes, I have contacted nvidia directly, to no avail.

I use https://github.com/lich426/FanCtrl to control my fan speeds. Yesterday, the app was updated with support for newer LHM libs, including the new GPU monitoring support. I immediately noticed that this version exhibits behaviour I have seen in other monitoring apps such as hwinfo (but, for example, not MSI Afterburner, and until recently, not LHM)

Essentially, the use of this monitoring causes stutters in-game. The stutters are small, so high and stable framerates are required to observe it. Here's a screenshot showing the game (in this case BF1 on a 3090 with a 162FPS in-game cap) running, then I alt-tab out and untick the 'GPU' box in LHM and alt-tab back in:
[image]

Using fanctrl, I can select between the in-app GPU monitoring, which works fine, and latest LHM, which is a guaranteed frametime spike with every query. Going back I can see the issue in 0.88 but not 0.87.

PhyxionNL, 2022-01-13:
Probably caused by some API call, which need NVIDIA fix.

xcasxcursex, 2022-06-22:
> Probably caused by some API call, which need NVIDIA fix.

Nvidia are completely unresponsive on this matter, perhaps you have some kind of contact there which you could leverage? Or maybe you could ask yourself and they won't ignore you because you're 'somebody'.

I'm surprised that "nobody with an Nvidia GPU can monitor VRAM or hotspot temps without a massive performance impact" isn't getting more traction, but here we are. HWInfo implemented a fix but they're closed source so who knows how they did it.