LibreHardwareMonitor/LibreHardwareMonitor #602: Feature Request - Fan Curve Control

Yahfz, 2021-11-13:
This software is amazing and it does nearly everything I want it to do except fan curves. While having the option of a fixed fan curve is excellent, having a fully customizable fan curve is even better.

ntoff, 2021-11-13:
Check out https://github.com/Rem0o/FanControl.Releases , it uses librehwmonitor's library so if your hardware is supported here, fancontrol most likely also supports it.

peacepenguin, 2021-11-15:
I've started trying to figure out this enhancement, so I created a basic proof of concept console app using the LHM library to do fan curve control with multiple input sensors, which works well:

https://github.com/peacepenguin/fancurver/

All the code is in this single file: https://github.com/peacepenguin/fancurver/blob/main/fancurver/Program.cs

I think this is representative of the absolute minimum code for a working fan curve control system.  All variables are hard-coded, so it makes it clumsy for a user to use, but is easy to read the code itself, which was my only goal here.

I'm now looking at where I can integrate these working concepts into LibreHardwareMonitor's UI.

Assuming its easy to make a new ui Form to select the input sensors to use, and define the Temp:Speed x,y coords, I'm trying to figure out where the curve calculation code would run. It looks like duplicating the Tray icon paradigm might work, where the updates of the tray icon only get called on sensors stored in the _settings that contain "tray". If a fan is marked as controlled by a curve, it could enter a new code block that asks for the current needed sensor data, do the SUM/MAX math, then apply the line-slope math to figure out the exact speed to set at that moment, and then actually set it to the controlled object.

peacepenguin, 2021-11-16:
I've now successfully integrated the poc 'fancurver' project directly into LibreHardwareMonitor, it works fine, but is only configurable at compile time, and is NOT efficient. It iterates through all sensors to find the defined inputs and outputs every refresh cycle.

But the important thing is that it works, and proves we CAN do this! Just need to improve efficiency with a list or something like the gadget and tray icon logic use, then make all the variables configurable at run time, then the GUI that can present all the complexity of a fan curve with multiple inputs, and how to assign each fan to a specific curve, or multiple curves.

Here's the fork where you can find the functional work in progress:

https://github.com/peacepenguin/LibreHardwareMonitor

Anyone wanting to experiment with the fan curve features so far or help continue the feature's progress:

Get visual studio community 2022:
https://visualstudio.microsoft.com/thank-you-downloading-visual-studio/?sku=Community&rel=17

When prompted, Install the "Workload" for ".net Desktop Development"

Then install the .net 452 dev pack:
https://dotnet.microsoft.com/download/dotnet-framework/net452

Then clone the repo within visual studio or within git bash for windows, and open the .sln file and then right click the Project "LibreHardwareMonitor" and click "build".

You'll want to adjust the fan inputs and outputs manually in the MainForm.cs  starting at line 67. Lots of hard coded variables to play with to get the curve behavior you're after.