LibreHardwareMonitor/LibreHardwareMonitor #614: [FanControl] Can't control fans on ASUS TUF B450M-PRO GAMING mobo

Anbeeld, 2021-12-10:
Hello. I have a ASUS TUF B450M-PRO GAMING motherboard and FanControl tool can't do anything with any of fans connected to it (while GPU fans are fine). Changing speeds do absolutely nothing. There are RPM values, but they are obviously incorrect with 1 fan showing something like 27 RPM and 2 others showing something around 3000 RPM, while in fact all of them are at 500-600 RPM.

Which should be an issue of LibreHardwareMonitor not supporting this motherboard. Is it possible to fix this?

For instance, HWiNFO64 shows RPM values of all the fans correctly.

kccricket, 2022-01-07:
I have the same motherboard and am (was) having the same issue, almost exactly as @Anbeeld described it. Libre Hardware Monitor and FanControl report chassis fan speeds much higher than the fan can spin (>3000 RPM) and the CPU fan reports a solid 21 RPM. Using the fan controls within LHM causes the fan speeds to *drop*, even when set to 100%. In this state, OpenHardwareMonitor doesn't even report any sensors under the ITE IT8655E component.

I've found that I can fix this state by quitting LHM and/or FanControl (if running), starting CPUID's HWMonitor, and then starting LHM or FanControl. I don't have to keep HWMonitor running or even have it running when I start LHM. I've tested this theory 8 different ways. It's as if HWMonitor is doing something to "initialize" the ITE IT8655E component that makes it good until the next boot.

This could be speculation, but I noticed this behavior immediately after installing AMD's chipset software version 3.10.22.706. However, neither downgrading to the version I had prior, 2.11.26.106, nor starting with a fresh install of Windows 10 fixed the issue.

If it's helpful, I've included Libre Hardware Monitor reports from when it's in the problematic and good states.

[LibreHardwareMonitor.Report Bad State](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/7825343/LibreHardwareMonitor.Report-kccricket-20220106.txt)

[LibreHardwareMonitor.Report Good State](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/7826198/LibreHardwareMonitor.Report-kccricket-20220106-good.txt)

Anbeeld, 2022-01-07:
> I have the same motherboard and am (was) having the same issue, almost exactly as @Anbeeld described it. Libre Hardware Monitor and FanControl report chassis fan speeds much higher than the fan can [...]

Oh my, this legit works. Started HWMonitor, started FanControl, boom! RPMs are real and I can control fans. Closed FanControl, closed HWMonitor, started FanControl. And still works. Huh. I'm on 2.11 chipset driver, just in case.

PhyxionNL, 2022-01-11:
Related: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/349

Shadow091995, 2022-01-14:
Same issue for me with an ASUS Prime B350-Plus with reported speeds of up to 12000RPM.
The workaround with HWMonitor works fine but it's a bit dumb to do it every time you restart imho.

It worked fine until a few weeks/months ago for me. Not sure what happened.

Kempes7475, 2022-02-27:
I have ASUS B450 TUF Gaming Plus II - the same problem. I use CPU-Z to "activate" correct fan speed. It would be great to avoid this step :) Maybe next version...

kccricket, 2022-03-24:
Oddly, this has started to occur after waking from sleep states, though there's been no change to my system BIOS or mainboard drivers. Since this bug may never be fixed, I'll share my automatic workaround. I'm using two Windows Scheduled Tasks: one to temporarily launch HWMonitor after boot and waking from sleep, and another to launch or re-launch FanSpeed at boot and waking from sleep. I'm attaching an archive containing exports of the two tasks. You'll need to edit both to insert your username and edit FanControl.xml to your path to FanControl.exe before importing them into Task Scheduler.

[ASUS-FanControl-Workaround.zip](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/8343563/ASUS-FanControl-Workaround.zip)

ErisaFirehawk, 2022-04-23:
I have an ASUS TUF x570 board and am having this same issue, but the work around stated here does seem to work for me.

ErisaFirehawk, 2022-04-23:
> I have an ASUS TUF x570 board and am having this same issue, but the work around stated here doesnt seem to work for me.

edited... big difference between does and doesn't

Svek, 2022-06-17:
A quick update... I've added some code to get into this issue further.
Some triage and a bit more testing might be needed. (https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/758)

simplebunsen, 2023-04-03:
Same Problem on my asus b350-prime.

igorrr12, 2023-08-27:
Same problem on Asus b450m-a II

BBarfryyy, 2023-10-19:
Hi, juste saying that it work for me. FanControl used to work perfectly but recently it didnt, i was looking for a solution when I saw this post. Im on asus b450 prime K II . This post / idea should be put forward. Thanks you !

flappyjet, 2024-03-08:
I set HWMonitor to win10 startup and FanControl delay 20 seconds. It does work for ASUS B450m PRO S. Great!

PR #1438 (2024-08-22): Bugfix: Fan and Temp data missing #780
This is a fix for #780 IT8655E which shows incorrect/missing information for fans and temps.  Also fixes #614.

I have tested this on my ASUS Prime B350M-A, through a number of restarts, sleeps and hibernates, and I no longer see the incorrect fan speeds, incorrect temperatures or missing temperatures.

Inspiration came from [Linux it87](https://github.com/frankcrawford/it87/blob/14b1de2fb49db4e73b58177bc43daa71ceb8e92f/it87.c#L3870) where the interaction with the IT8655E does a set_bank before every interaction with a port.

In theory this fix can be app [... cut]
closed 2024-08-22 by PhyxionNL