LibreHardwareMonitor/LibreHardwareMonitor #636: Multiple GPUs showing exact same memory (free,used,total) ... despite not all having 8GB onboard.

roboyeti, 2022-01-22:
Issue:  In a system with multiple cards, the memory free,used,and total values are set for all cards to the values of the first card in the bus.  I know this is true because the data below shows correctly for an RTX 2060 Super w/8GB, but shows the GTX 1660 Supers with 6GB ram as having 8GB total, as well as the values being otherwise suspicious in absolute identical value.

Sensor data showing the issue:
{"Parent"=>"/gpu-nvidia/0",
  "Identifier"=>"/gpu-nvidia/0/smalldata/0",
  "InstanceId"=>"3877",
  "Name"=>"GPU Memory Free",
  "Value"=>3169,
  "SensorType"=>"SmallData"},
 {"Parent"=>"/gpu-nvidia/0",
  "Identifier"=>"/gpu-nvidia/0/smalldata/1",
  "InstanceId"=>"3878",
  "Name"=>"GPU Memory Used",
  "Value"=>5022,
  "SensorType"=>"SmallData"},
 {"Parent"=>"/gpu-nvidia/0",
  "Identifier"=>"/gpu-nvidia/0/smalldata/2",
  "InstanceId"=>"3876",
  "Name"=>"GPU Memory Total",
  "Value"=>8192,
  "SensorType"=>"SmallData"},
 {"Parent"=>"/gpu-nvidia/1",
  "Identifier"=>"/gpu-nvidia/1/smalldata/1",
  "InstanceId"=>"3896",
  "Name"=>"GPU Memory Free",
  "Value"=>3169,
  "SensorType"=>"SmallData"},
 {"Parent"=>"/gpu-nvidia/1",
  "Identifier"=>"/gpu-nvidia/1/smalldata/1",
  "InstanceId"=>"3897",
  "Name"=>"GPU Memory Used",
  "Value"=>5022,
  "SensorType"=>"SmallData"},
 {"Parent"=>"/gpu-nvidia/1",
  "Identifier"=>"/gpu-nvidia/1/smalldata/2",
  "InstanceId"=>"3895",
  "Name"=>"GPU Memory Total",
  "Value"=>8192,
  "SensorType"=>"SmallData"},
 {"Parent"=>"/gpu-nvidia/2",
  "Identifier"=>"/gpu-nvidia/2/smalldata/0",
  "InstanceId"=>"3915",
  "Name"=>"GPU Memory Free",
  "Value"=>3169,
  "SensorType"=>"SmallData"},
 {"Parent"=>"/gpu-nvidia/2",
  "Identifier"=>"/gpu-nvidia/2/smalldata/1",
  "InstanceId"=>"3916",
  "Name"=>"GPU Memory Used",
  "Value"=>5022,
  "SensorType"=>"SmallData"},
 {"Parent"=>"/gpu-nvidia/2",
  "Identifier"=>"/gpu-nvidia/2/smalldata/2",
  "InstanceId"=>"3914",
  "Name"=>"GPU Memory Total",
  "Value"=>8192,
  "SensorType"=>"SmallData"},

Image of interface confirms it:
[image]

rik43, 2024-11-29:
Same issue, but with VRAM memory info. 
i have:
rtx 3080ti 20gb
rtx 3090 24gb
and see memory free/used/total from first (3080ti). 
3090 used 23gb of 24gb, but its show 19gb is free. 

[image]

rik43, 2024-11-29:
Same problem in open hardware monitor