LibreHardwareMonitor/LibreHardwareMonitor #639: GetPmTableStructure returns empty Dictionary

DevTechProfile, 2022-01-22:
On my system (Ryzen 5900X) GetPmTableStructure returns an empty Dictionary. Supported PM table versions are
[image]

but the current PM table version is 3672069. Therefore not SMU sensors can be read out.

DevTechProfile, 2022-01-22:
Sorry, fixed it quickly by adding the PM table version to the dictionary.

`private readonly Dictionary<uint, Dictionary<uint, SmuSensorType>> _supportedPmTableVersions = new()
        {
            {
                // Zen Raven Ridge APU.
                0x001E0004, new Dictionary<uint, SmuSensorType>
                {
                    { 7, new SmuSensorType { Name = "TDC", Type = SensorType.Current, Scale = 1 } },
                    { 11, new SmuSensorType { Name = "EDC", Type = SensorType.Current, Scale = 1 } },
                    //{ 61, new SmuSensorType { Name = "Core", Type = SensorType.Voltage } },
                    //{ 62, new SmuSensorType { Name = "Core", Type = SensorType.Current, Scale = 1} },
                    //{ 63, new SmuSensorType { Name = "Core", Type = SensorType.Power, Scale = 1 } },
                    //{ 65, new SmuSensorType { Name = "SoC", Type = SensorType.Voltage } },
                    { 66, new SmuSensorType { Name = "SoC", Type = SensorType.Current, Scale = 1 } },
                    { 67, new SmuSensorType { Name = "SoC", Type = SensorType.Power, Scale = 1 } },
                    //{ 96, new SmuSensorType { Name = "Core #1", Type = SensorType.Power } },
                    //{ 97, new SmuSensorType { Name = "Core #2", Type = SensorType.Power } },
                    //{ 98, new SmuSensorType { Name = "Core #3", Type = SensorType.Power } },
                    //{ 99, new SmuSensorType { Name = "Core #4", Type = SensorType.Power } },
                    { 108, new SmuSensorType { Name = "Core #1", Type = SensorType.Temperature, Scale = 1 } },
                    { 109, new SmuSensorType { Name = "Core #2", Type = SensorType.Temperature, Scale = 1 } },
                    { 110, new SmuSensorType { Name = "Core #3", Type = SensorType.Temperature, Scale = 1 } },
                    { 111, new SmuSensorType { Name = "Core #4", Type = SensorType.Temperature, Scale = 1 } },
                    { 150, new SmuSensorType { Name = "GFX", Type = SensorType.Voltage, Scale = 1 } },
                    { 151, new SmuSensorType { Name = "GFX", Type = SensorType.Temperature, Scale = 1 } },
                    { 154, new SmuSensorType { Name = "GFX", Type = SensorType.Clock, Scale = 1 } },
                    { 156, new SmuSensorType { Name = "GFX", Type = SensorType.Load, Scale = 1 } },
                    { 166, new SmuSensorType { Name = "Fabric", Type = SensorType.Clock, Scale = 1 } },
                    { 177, new SmuSensorType { Name = "Uncore", Type = SensorType.Clock, Scale = 1 } },
                    { 178, new SmuSensorType { Name = "Memory", Type = SensorType.Clock, Scale = 1 } },
                    { 342, new SmuSensorType { Name = "Displays", Type = SensorType.Factor, Scale = 1 } },
                }
            },
            {
                // Zen 2.
                0x00240903, new Dictionary<uint, SmuSensorType>
                {
                    { 15, new SmuSensorType { Name = "TDC", Type = SensorType.Current, Scale = 1 } },
                    { 21, new SmuSensorType { Name = "EDC", Type = SensorType.Current, Scale = 1 } },
                    { 48, new SmuSensorType { Name = "Fabric", Type = SensorType.Clock, Scale = 1 } },
                    { 50, new SmuSensorType { Name = "Uncore", Type = SensorType.Clock, Scale = 1 } },
                    { 51, new SmuSensorType { Name = "Memory", Type = SensorType.Clock, Scale = 1 } },
                    { 115, new SmuSensorType { Name = "SoC", Type = SensorType.Temperature, Scale = 1 } },
                    //{ 66, new SmuSensorType { Name = "Bus Speed", Type = SensorType.Clock, Scale = 1 } },
                    //{ 188, new SmuSensorType { Name = "Core #1", Type = SensorType.Clock, Scale = 1000 } },
                    //{ 189, new SmuSensorType { Name = "Core #2", Type = SensorType.Clock, Scale = 1000 } },
                    //{ 190, new SmuSensorType { Name = "Core #3", Type = SensorType.Clock, Scale = 1000 } },
                    //{ 191, new SmuSensorType { Name = "Core #4", Type = SensorType.Clock, Scale = 1000 } },
                    //{ 192, new SmuSensorType { Name = "Core #5", Type = SensorType.Clock, Scale = 1000 } },
                    //{ 193, new SmuSensorType { Name = "Core #6", Type = SensorType.Clock, Scale = 1000 } },
                }
            },
            {
                // Zen 3 
                0x00380805,
                new Dictionary<uint, SmuSensorType>
                {
                    { 15, new SmuSensorType { Name = "TDC", Type = SensorType.Current, Scale = 1 } },
                    { 21, new SmuSensorType { Name = "EDC", Type = SensorType.Current, Scale = 1 } },
                    { 48, new SmuSensorType { Name = "Fabric", Type = SensorType.Clock, Scale = 1 } },
                    { 50, new SmuSensorType { Name = "Uncore", Type = SensorType.Clock, Scale = 1 } },
                    { 51, new SmuSensorType { Name = "Memory", Type = SensorType.Clock, Scale = 1 } },
                    { 115, new SmuSensorType { Name = "SoC", Type = SensorType.Temperature, Scale = 1 } },
                    //{ 66, new SmuSensorType { Name = "Bus Speed", Type = SensorType.Clock, Scale = 1 } },
                    //{ 188, new SmuSensorType { Name = "Core #1", Type = SensorType.Clock, Scale = 1000 } },
                    //{ 189, new SmuSensorType { Name = "Core #2", Type = SensorType.Clock, Scale = 1000 } },
                    //{ 190, new SmuSensorType { Name = "Core #3", Type = SensorType.Clock, Scale = 1000 } },
                    //{ 191, new SmuSensorType { Name = "Core #4", Type = SensorType.Clock, Scale = 1000 } },
                    //{ 192, new SmuSensorType { Name = "Core #5", Type = SensorType.Clock, Scale = 1000 } },
                    //{ 193, new SmuSensorType { Name = "Core #6", Type = SensorType.Clock, Scale = 1000 } },
                }
            }
        };
`

PhyxionNL, 2022-01-23:
Feel free to send a PR with this code.

DevTechProfile, 2022-01-24:
Created PR

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/640

closed 2022-01-24 by DevTechProfile