LibreHardwareMonitor/LibreHardwareMonitor #642: NvApi.dll on linux

Azaran, 2022-01-26:
Hey,  I'm playing around with `LibreHardwareMonitorLib` on my CentOS 7 server. The server is equiped with several Nvidia RTX 2080Ti which I want to monitor using my own C# app with added the lib Nugget package. CPU and other things get discovered correctly, but GPUs are not getting recognize and code inconstructor of `NvApi()` class fails with `DllNotFoundException`. 

How did you get `nvapi.dll` or `nvapi64.dll` on Linux? Aren't you trying to load Windows specific libraries on Linux which causes the issue? I'm asking because I can't find any NvApi implementation or installer for Linux. Any ideas how to fix the GPU detection on Linux?

Tederean, 2022-04-15:
NVAPI is only available on windows and thus LibreHardwareMonitor does not support Nvidia GPUs under Linux either.

But that doesn't have to be the case, since most of the information is also available under Linux with the NVML API. I'm working on a patch for this.

Azaran, 2022-04-26:
Cool, that kinda exactly what I did in my product. I've ditched NVAPI and gathered the info I've needed from NVML API. But I don't have Windows installation with NVIDIA, so I couldn't test it and PR out of it. Good luck with the patch, I'm looking forward to potentially migration to your solution as I suppose it will be more generic than mine.

Alien-hide, 2025-01-27:
@Tederean 
How is your progress going? When do you expect to have it ready?
Thanks!