LibreHardwareMonitor/LibreHardwareMonitor #643: Support for Aquacomputer sensors

AstralStorm, 2022-01-28:
Support for Aquacomputer hardware would be nice.
There's a long list of these and they seem to use the same protocol.

It seems it's already supported in HWInfo and AIDA64, and likely also possible to handle in LibreHardwareMonitor after some communication with the company.

AstralStorm, 2022-01-28:
These are USB HID devices, and there exists support for Aquacomputer Octa and Quadro in form of Linux kernel drivers. Shouldn't be too hard to actually support other devices.

eikaramba, 2023-03-13:
d5 next is already there ;) why i know that? because i added it to this project.

Jay2k1, 2025-01-05:
I have a poweradjust 3.0 which is not yet supported (device ID `f0bd`). Here are the sensors that can be read:
[image]

It would be great if someone could have a go at adding support for it - I can provide more information on the device if necessary and try test builds.