LibreHardwareMonitor/LibreHardwareMonitor #645: AMD Virtualization (SVM) Causes Clock Speeds to Report Incorrectly

znbaboy, 2022-02-02:
https://github.com/ArcadeRenegade/SidebarDiagnostics/issues/388

nickolasdeluca, 2022-05-01:
Any fix on this issue?

Smokierpizza17, 2022-05-09:
I'm having pretty much the exact same issue. I can have SVM enabled in the BIOS, but it's when I turn on the Windows Hypervisor Platform and Virtual Maschine Platform that it doesn't work. I've attached a report, but here's the more important bit. It seems to change between ~15.2GHz, Infinity and NaN. My guess is it's something to do with the Hypervisor not sharing information on the clock speeds the same way as without. I'm already running LibreHardwareMonitor as admin.

```
...
|  +- Core #1        :      NaN  12.8786 Infinity (/amdcpu/0/clock/1)
|  +- Core #2        :      NaN        0 Infinity (/amdcpu/0/clock/2)
|  +- Core #3        :  37.4283      NaN      NaN (/amdcpu/0/clock/3)
|  +- Core #4        :      NaN  5.70336 Infinity (/amdcpu/0/clock/4)
|  +- Core #5        :  15270.8        0 Infinity (/amdcpu/0/clock/5)
|  +- Core #6        :      NaN  37.4283 Infinity (/amdcpu/0/clock/6)
...
```
[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/8654844/LibreHardwareMonitor.Report.txt)

LiloBzH, 2022-09-02:
same here

[image]

DjordjeMandic, 2023-03-09:
Voltages are off also, both debug and release build 

[image]
[image]