LibreHardwareMonitor/LibreHardwareMonitor #649: Missing Drives

r3d1r, 2022-02-04:
In one computer I have 5 physical drives:
 - 2 Hard drives 
 - 2 SSDs
 - 1 NVMe M.2 SSD 

The first 4 are "joined" in a single Microsoft Storage Space virtual drive.

Libre Hardware monitor shows the NVMe sensors and the Msft Storage space data.
It should also show the hard disks and SSDs sensors, but it didn´t.

In the other hand, on Open Hardware ( original source of this fork ) I can see hard disks and SSDs sensors, but it misses the NVMe...

Images attached

[image]

[image]

Windows Device Manager

[image]
 
Regards,
Maurício Faria

r3d1r, 2022-02-04:
I'm not sure, but this can be relevant:
Main board is an Asus Prime B450M-Gaming/BR and it has an ITE IT8655E with an AMD Ryzen 5 2600X and its running a Windows Server 2019.

TowardtheStars, 2024-01-21:
I also encountered such problem.
## Device Manager
[image]
## Libre Hardware Monitor
[image]

System Info
--------
Open Hardware Monitor : 0.9.3
System: Windows 11 22635.2921
Motherboard: ASUS B650M TUF Gaming Plus D4

Disk Configuration
=================
Disk Name (from Device Manager) | Disk Type | Installation | Transfer Mode (from CrystalDiskInfo 8.17.14) |
-----------|---------|---------|------------|
CT2000MX500SSD1| 2.5' SSD | SATA port on motherboard | SATA/600|
Fanxiang S500Pro 2TB| SSD | Extended m.2 slot on PCIe extension card | PCIe 3.0 x1|
HGST HUS 728T8TALE6L4 SCSI Disk Device | HDD | SATA3 to USB3.0 box | SATA/600 |
HS-SSD-C4000ECO 2048G| SSD | Extended m.2 slot on PCIe to m.2 extension card | PCIe 3.0 x1|
Predator SSD GM7 M.2 2TB | SSD | Extended m.2 slot on PCIe to m.2 extension card | PCIe 3.0 x1|
Samsung SSD 970 EVO 1TB | SSD | 1st m.2 slot on motherboard | PCIe 3.0 x4|
WDC WD10SPZX-75Z10T2 | 2.5' HDD | SATA port  on motherboard| SATA/600|

closed 2025-12-30 by PhyxionNL