LibreHardwareMonitor/LibreHardwareMonitor #669: [Fan Control] No fan control on Gigabyte Z690 Aorus Elite DDR4

Zarco92, 2022-02-19:
Ever since upgrading to a 12700K + Gigabyte Z690 Aorus Elite DDR4, I haven't been able to control any fans connected to the motherboard via Fan Control. Fan control for my RTX 3080 works fine. Afaik, hardware compatibility issues are entirely dependant on these libraries so I'll report it here.

Running the latest release of Fan Conrol (103) with the latest LibreHardwareMonitor update., on Windows 11.

zokradonh, 2022-02-22:
Please post your LibreHardwareMonitor.Report.txt that can be generated with the LibreHardwareMonitor application.
I have the Gigabyte Z690 Aorus Pro DDR5 and also same problems. I am currently trying to fix it.

Zarco92, 2022-02-22:
Will do ASAP, thanks

zokradonh, 2022-02-22:
Ok, your mainboard also uses IT8689E like my mainboard. So it will work once I have finished #672

Zarco92, 2022-02-22:
Damn, that's great. Let me know if you need something else. Should I close this issue now? I'm kinda new to github.

zokradonh, 2022-02-22:
You can close this issue once it really works for you.

Zarco92, 2022-02-22:
Alright. Thanks again!

Zarco92, 2022-04-07:
Forgot to close this one, sorry about that.

closed 2022-04-07 by Zarco92