LibreHardwareMonitor/LibreHardwareMonitor #683: Read Rate and Write Rate

kensh1kk, 2022-03-01:
When I try to get Value by sensor.Value method from LibreHardwareMonitorLib.dll , everything is output well, except for the speed of writing and reading disks. 

A strange value is output that does not change when the disk is loaded. In the exe program, the information is displayed correctly.
For example: 

Read Rate 34 410728.5 /hdd/0/throughput/34
Write Rate 35 358923.5 /hdd/0/throughput/35

`

     import os
     import clr #pythonnet
    
    
    def initialize():
    
        clr.AddReference(os.getcwd() + "\\LibreHardwareMonitorLib.dll")
    
        from LibreHardwareMonitor import Hardware
    
        handle = Hardware.Computer()
        handle.IsCpuEnabled = True
        handle.IsGpuEnabled = True
        handle.IsMemoryEnabled = True
        handle.IsMotherboardEnabled = True
        handle.IsControllerEnabled = True
        handle.IsNetworkEnabled = True
        handle.IsStorageEnabled = True
        handle.Open()
        return handle
    
    
    def fetch_stats(handle):
        for i in handle.Hardware:
            i.Update()
            for sensor in i.Sensors:
                parse_sensor(sensor)
    
    
    def parse_sensor(sensor):
        if sensor.Value is not None:
            data = (f"{sensor.Hardware.Name} "
                         f"{sensor.Name} "
                         f"{sensor.Value}")
            print(data)
  
  
    if __name__ == "__main__":
        HardwareHandle = initialize()
        fetch_stats(HardwareHandle)
`
language python 3.8

ivanbogaeb, 2022-03-02:
Hey, I am working on the same thing as you but for a different project. What happens there is that you are not enabling the hardware sensors correctly, what I have done is this:

```python
  handle.set_IsStorageEnabled(True)
  handle.set_IsNetworkEnabled(True)
  handle.set_IsMemoryEnabled(True)
  handle.set_IsGpuEnabled(True)
  handle.set_IsCpuEnabled(True)
```

If you check inside the library, you are going to find out that the `set_IsStuff` functions enable the respective hardware and sensors.

kensh1kk, 2022-03-02:
Thanks a lot, it worked for me!

closed 2022-03-02 by kensh1kk

ivanbogaeb, 2022-03-02:
No problem! I recommend you to explore the `Hardware.Computer` attributes by using `Hardware.Computer.__dict__` , it will give you a lot of useful information on how to handle sensors, and subhardware!