LibreHardwareMonitor/LibreHardwareMonitor #687: [Fan Control] No fan control on motherboard HP 88D0 (HP 15 Pavilion enh1* with AMD Ryzen 5)

Jhunboss, 2022-03-06:
Hello,

Fans are not detected for my HP Pavilion 15 enh1026 with AMD Ryzen 5 5500U CPU. I guess my motherboard is HP 88D0.
Can you add it please or there is a way to do it himself ? 

Thank you very much !

Below my LibreHardwareMonitor.txt Report

--------------------------------------------------------------------------------

Version: 0.8.9.0

--------------------------------------------------------------------------------

Common Language Runtime: 4.0.30319.42000
Operating System: Microsoft Windows NT 10.0.22000.0
Process Type: 64-Bit

--------------------------------------------------------------------------------

Sensors

|
+- HP 88D0 (/motherboard)
|
+- AMD Ryzen 5 5500U with Radeon Graphics (/amdcpu/0)
|  +- Core (SVI2 TFN) :     1.55     1.55     1.55 (/amdcpu/0/voltage/0)
|  +- SoC (SVI2 TFN) :     1.55     1.55     1.55 (/amdcpu/0/voltage/1)
|  +- Core #1 VID    :  0.96875      0.5    1.175 (/amdcpu/0/voltage/2)
|  +- Core #2 VID    :  0.96875      0.5    1.175 (/amdcpu/0/voltage/3)
|  +- Core #3 VID    :  0.96875      0.5    1.175 (/amdcpu/0/voltage/4)
|  +- Core #4 VID    :  0.96875      0.5  1.16875 (/amdcpu/0/voltage/5)
|  +- Core #5 VID    :  0.96875      0.5  1.16875 (/amdcpu/0/voltage/6)
|  +- Core #6 VID    :  0.96875      0.5  1.16875 (/amdcpu/0/voltage/7)
|  +- Package        :  1.82177  1.54898  9.41816 (/amdcpu/0/power/0)
|  +- Core #1 (SMU)  : 0.0592854 0.0264915  1.16153 (/amdcpu/0/power/1)
|  +- Core #2 (SMU)  : 0.0415074 0.0362507  1.01994 (/amdcpu/0/power/2)
|  +- Core #3 (SMU)  : 0.0436079 0.0161385 0.654651 (/amdcpu/0/power/3)
|  +- Core #4 (SMU)  : 0.0510357 0.0250147  1.04114 (/amdcpu/0/power/4)
|  +- Core #5 (SMU)  : 0.0433796 0.021924  1.33057 (/amdcpu/0/power/5)
|  +- Core #6 (SMU)  : 0.142559 0.0619559  4.02262 (/amdcpu/0/power/6)
|  +- Bus Speed      :  99.8125  99.8125  99.8129 (/amdcpu/0/clock/0)
|  +- Core #1        :  2071.11  1924.96  4042.42 (/amdcpu/0/clock/1)
|  +- Core #2        :  2071.11  1924.96  4042.41 (/amdcpu/0/clock/2)
|  +- Core #3        :  2071.11  1924.96  4042.41 (/amdcpu/0/clock/3)
|  +- Core #4        :  2071.11  1924.96  4042.41 (/amdcpu/0/clock/4)
|  +- Core #5        :  2071.11  1924.95  4042.41 (/amdcpu/0/clock/5)
|  +- Core #6        :  2071.11  1924.95  4042.41 (/amdcpu/0/clock/6)
|  +- Core (Tctl/Tdie) :     48.5     48.5    53.25 (/amdcpu/0/temperature/2)
|  +- CPU Total      :  2.07143 0.651044  10.1436 (/amdcpu/0/load/0)
|  +- CPU Core #1    :     6.25        0      100 (/amdcpu/0/load/1)
|  +- CPU Core #2    :  1.53846        0    3.125 (/amdcpu/0/load/2)
|  +- CPU Core #3    :  1.53846        0       25 (/amdcpu/0/load/3)
|  +- CPU Core #4    :        0        0       20 (/amdcpu/0/load/4)
|  +- CPU Core #5    :    3.125        0       20 (/amdcpu/0/load/5)
|  +- CPU Core #6    :   1.5625        0  4.76191 (/amdcpu/0/load/6)
|  +- CPU Core #7    :   3.0303        0       40 (/amdcpu/0/load/7)
|  +- CPU Core #8    :   1.5625        0  33.3333 (/amdcpu/0/load/8)
|  +- CPU Core #9    :        0        0  33.3333 (/amdcpu/0/load/9)
|  +- CPU Core #10   :        0        0  9.30232 (/amdcpu/0/load/10)
|  +- CPU Core #11   :    3.125        0  66.6667 (/amdcpu/0/load/11)
|  +- CPU Core #12   :    3.125        0  4.65117 (/amdcpu/0/load/12)
|  +- Core #1        :    20.75  19.2857     40.5 (/amdcpu/0/factor/0)
|  +- Core #2        :    20.75  19.2857     40.5 (/amdcpu/0/factor/1)
|  +- Core #3        :    20.75  19.2857     40.5 (/amdcpu/0/factor/2)
|  +- Core #4        :    20.75  19.2857     40.5 (/amdcpu/0/factor/3)
|  +- Core #5        :    20.75  19.2857     40.5 (/amdcpu/0/factor/4)
|  +- Core #6        :    20.75  19.2857     40.5 (/amdcpu/0/factor/5)
|
[... 1280 lines cut ...]

Network Utilization
0

Realtek RTL8852AE WiFi 6 802.11ax PCIe Adapter
Up

Data Uploaded
0,1080935

Data Downloaded
7,528178

Upload Speed
54,69432

Download Speed
65,63319

Network Utilization
0,0001110673

--------------------------------------------------------------------------------

Corsair HXi/RMi series PSU Hardware

azzydoesgit, 2022-05-04:
Dude holy shit, use a pastebin or something

Jhunboss, 2022-05-07:
Sorry i’m new on Github

Jhunboss, 2022-05-07:
I'm new on github. You can find a pastebin here : https://pastebin.com/rpGkzwXm.
Thanks

azzydoesgit, 2022-05-07:
> I'm new on github. You can find a pastebin here : https://pastebin.com/rpGkzwXm. Thanks

It's all good, it's just that the plaintext is _super_ long

Jhunboss, 2022-05-07:
I know, I compared with other requests and I saw that some did like that. I thought that was the correct procedure.
Anyway, let me know if I can help with my request.