LibreHardwareMonitor/LibreHardwareMonitor #688: Index out of Range issue in TreeViewAdv.cs after returning from sleep

Eno77, 2022-03-07:
I use the latest build from 1. March (pre97):

The following Error is shown, but LHM runs normal after that.
I hope the Error Text gives a hint to the issue.

************** Ausnahmetext **************
System.ArgumentOutOfRangeException: Das angegebene Argument liegt außerhalb des gültigen Wertebereichs.
Parametername: Index out of range
   bei Aga.Controls.Tree.TreeViewAdv.ClearNodesSize(TreeModelEventArgs e, TreeNodeAdv parent) in D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeViewAdv.cs:Zeile 1227.

mindplay-dk, 2023-06-28:
I got a very similar error not after hibernation, but just after cold booting my computer this morning.

Here's the full stack trace:

```
See the end of this message for details on invoking 
just-in-time (JIT) debugging instead of this dialog box.

************** Exception Text **************
System.ArgumentOutOfRangeException: Index was out of range. Must be non-negative and less than the size of the collection.
Parameter name: index
   at System.ThrowHelper.ThrowArgumentOutOfRangeException(ExceptionArgument argument, ExceptionResource resource)
   at System.Collections.Generic.List`1.get_Item(Int32 index)
   at Aga.Controls.Tree.TreeNodeAdv.get_NextNode() in D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeNodeAdv.cs:line 252
   at Aga.Controls.Tree.TreeNodeAdv.get_NextVisibleNode() in D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeNodeAdv.cs:line 281
   at Aga.Controls.Tree.TreeViewAdv.<get_VisibleNodes>d__256.MoveNext() in D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeViewAdv.Properties.cs:line 74
   at Aga.Controls.Tree.TreeViewAdv.CreateRowMap() in D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeViewAdv.cs:line 890
   at Aga.Controls.Tree.TreeViewAdv.UnsafeFullUpdate() in D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeViewAdv.cs:line 684
   at System.Windows.Forms.Control.InvokeMarshaledCallbackHelper(Object obj)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Windows.Forms.Control.InvokeMarshaledCallback(ThreadMethodEntry tme)
   at System.Windows.Forms.Control.InvokeMarshaledCallbacks()

************** Loaded Assemblies **************
mscorlib
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4644.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
LibreHardwareMonitor
    Assembly Version: 0.9.2.0
    Win32 Version: 0.9.2.0
    CodeBase: file:///C:/Program%20Files/Libre%20Hardware%20Monitor/LibreHardwareMonitor.exe
----------------------------------------
System.Windows.Forms
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4550.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4644.0 built by: NET48REL1LAST_B
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4390.0 built by: NET48REL1LAST_C
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
Aga.Controls
    Assembly Version: 1.7.0.0
    Win32 Version: 1.7.0.0
    CodeBase: file:///C:/Program%20Files/Libre%20Hardware%20Monitor/Aga.Controls.DLL
----------------------------------------
LibreHardwareMonitorLib
    Assembly Version: 0.9.2.0
    Win32 Version: 0.9.2.0
    CodeBase: file:///C:/Program%20Files/Libre%20Hardware%20Monitor/LibreHardwareMonitorLib.DLL
----------------------------------------
Microsoft.Win32.TaskScheduler
    Assembly Version: 2.10.1.0
    Win32 Version: 2.10.1.0
    CodeBase: file:///C:/Program%20Files/Libre%20Hardware%20Monitor/Microsoft.Win32.TaskScheduler.DLL
----------------------------------------
System.Xml
    Assembly Version: 4.0.0.0
    Win32 Version: 4.8.4084.0 built by: NET48REL1
    CodeBase: file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll
[... 35 lines cut ...]
----------------------------------------
HidSharp
    Assembly Version: 2.1.0.0
    Win32 Version: 2.1.0.0
    CodeBase: file:///C:/Program%20Files/Libre%20Hardware%20Monitor/HidSharp.DLL
----------------------------------------

************** JIT Debugging **************
To enable just-in-time (JIT) debugging, the .config file for this
application or computer (machine.config) must have the
jitDebugging value set in the system.windows.forms section.
The application must also be compiled with debugging
enabled.

For example:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

When JIT debugging is enabled, any unhandled exception
will be sent to the JIT debugger registered on the computer
rather than be handled by this dialog box.

```

sintech, 2024-03-28:
I observe similar error while playing Death Stranding
```
System.ArgumentOutOfRangeException: Заданный аргумент находится вне диапазона допустимых значений.
Имя параметра: Index out of range
   в Aga.Controls.Tree.TreeViewAdv.ClearNodesSize(TreeModelEventArgs e, TreeNodeAdv parent) в D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeViewAdv.cs:строка 1227
```

Z3r10, 2024-10-30:
Just for reference, I opened another issue for this (sorry for the duplicate): https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/1506

On my machine it is random, and not related to sleep nor hibernation (i just got it this morning 1 minute after cold boot).

klinki, 2025-02-21:
I'm also hitting this.

<details><summary>Details</summary>

```
Informace o vyvolání ladění JIT najdete na konci této zprávy,
nikoli v tomto dialogovém okně.

************** Text výjimky **************
System.ArgumentOutOfRangeException: Index je mimo rozsah. Index musí být nezáporný a musí být menší než velikost kolekce.
Název parametru: index
   v System.ThrowHelper.ThrowArgumentOutOfRangeException(ExceptionArgument argument, ExceptionResource resource)
   v System.Collections.Generic.List`1.get_Item(Int32 index)
   v Aga.Controls.Tree.TreeNodeAdv.get_NextNode() v D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeNodeAdv.cs:řádek 252
   v Aga.Controls.Tree.TreeNodeAdv.get_NextVisibleNode() v D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeNodeAdv.cs:řádek 281
   v Aga.Controls.Tree.TreeViewAdv.<get_VisibleNodes>d__261.MoveNext() v D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeViewAdv.Properties.cs:řádek 74
   v Aga.Controls.Tree.TreeViewAdv.CreateRowMap() v D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeViewAdv.cs:řádek 900
   v Aga.Controls.Tree.TreeViewAdv.UnsafeFullUpdate() v D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeViewAdv.cs:řádek 694
   v System.Windows.Forms.Control.InvokeMarshaledCallbackHelper(Object obj)
   v System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   v System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   v System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   v System.Windows.Forms.Control.InvokeMarshaledCallback(ThreadMethodEntry tme)
   v System.Windows.Forms.Control.InvokeMarshaledCallbacks()

************** Načtená sestavení **************
mscorlib
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9290.0 built by: NET481REL1LAST_C
    Základ kódu (CodeBase): file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll
----------------------------------------
LibreHardwareMonitor
    Verze sestavení: 0.9.4.0
    Verze Win32: 0.9.4.0
    Základ kódu (CodeBase): file:///C:/Users/david/AppData/Local/Microsoft/WinGet/Packages/LibreHardwareMonitor.LibreHardwareMonitor_Microsoft.Winget.Source_8wekyb3d8bbwe/LibreHardwareMonitor.exe
----------------------------------------
System.Windows.Forms
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9256.0 built by: NET481REL1LAST_B
    Základ kódu (CodeBase): file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll
----------------------------------------
System
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9282.0 built by: NET481REL1LAST_C
    Základ kódu (CodeBase): file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll
----------------------------------------
System.Drawing
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9032.0 built by: NET481REL1
    Základ kódu (CodeBase): file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll
----------------------------------------
Aga.Controls
    Verze sestavení: 1.7.0.0
    Verze Win32: 1.7.0.0
    Základ kódu (CodeBase): file:///C:/Users/david/AppData/Local/Microsoft/WinGet/Packages/LibreHardwareMonitor.LibreHardwareMonitor_Microsoft.Winget.Source_8wekyb3d8bbwe/Aga.Controls.DLL
----------------------------------------
LibreHardwareMonitorLib
    Verze sestavení: 0.9.4.0
    Verze Win32: 0.9.4.0
    Základ kódu (CodeBase): file:///C:/Users/david/AppData/Local/Microsoft/WinGet/Packages/LibreHardwareMonitor.LibreHardwareMonitor_Microsoft.Winget.Source_8wekyb3d8bbwe/LibreHardwareMonitorLib.DLL
----------------------------------------
System.Core
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9277.0 built by: NET481REL1LAST_B
    Základ kódu (CodeBase): file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll
----------------------------------------
Microsoft.Win32.TaskScheduler
    Verze sestavení: 2.11.0.0
    Verze Win32: 2.11.0.0
    Základ kódu (CodeBase): file:///C:/Users/david/AppData/Local/Microsoft/WinGet/Packages/LibreHardwareMonitor.LibreHardwareMonitor_Microsoft.Winget.Source_8wekyb3d8bbwe/Microsoft.Win32.TaskScheduler.DLL
[... 45 lines cut ...]
----------------------------------------
System.Windows.Forms.resources
    Verze sestavení: 4.0.0.0
    Verze Win32: 4.8.9032.0 built by: NET481REL1
    Základ kódu (CodeBase): file:///C:/WINDOWS/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.resources/v4.0_4.0.0.0_cs_b77a5c561934e089/System.Windows.Forms.resources.dll
----------------------------------------

************** Ladění JIT **************
Aby bylo povoleno ladění JIT, musí konfigurační soubor
pro tuto aplikaci nebo počítač (machine.config) mít
v oddílu system.windows.forms nastavenou hodnotu njitDebugging.
Aplikace rovněž musí být kompilována s povoleným
laděním.

Příklad:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

Je-li ladění JIT povoleno, budou všechny neošetřené výjimky
odeslány ladicímu programu JIT, který je registrován v počítači,
a nebudou zpracovány tímto dialogovým oknem.
```
</details>

MayankJairaj, 2025-03-19:
For now I have just skipped the exception, will try and find some solution.

```
if (index < 0 || index >= parent.Nodes.Count)
{
continue;
}

```

scardus, 2026-09-30:
I hit this on my media server today (`CreateRowMap`, line 900) and had a dig into it.  I think it's a threading issue rather than anything specific to sleep.

`MainForm.HardwareAdded` and `HardwareRemoved` add and remove tree nodes on whatever thread raised the event, not the UI thread:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/677a3a56abde9adff5abdb42db3a7638c9137572/LibreHardwareMonitor.Windows.Forms/UI/MainForm.cs#L834-L855

The tree sees it's off the UI thread and queues its rebuild with `BeginInvoke`:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/677a3a56abde9adff5abdb42db3a7638c9137572/Aga.Controls/Tree/TreeViewAdv.cs#L681-L688

So the UI thread can end up walking the nodes while another thread is still changing them, and read past the end of a list.  My trace went through `InvokeMarshaledCallbacks` into `UnsafeFullUpdate`, which only happens when the tree was changed from another thread.  The different traces here (`ClearNodesSize`, `CreateRowMap`, `NextNode`, `BottomNode`) all look like the tree being changed off the UI thread, just caught at different points.

A few things change the tree off the UI thread:

- `NetworkGroup`, from `NetworkChange.NetworkAddressChanged`, whenever an adapter comes or goes: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/677a3a56abde9adff5abdb42db3a7638c9137572/LibreHardwareMonitorLib/Hardware/Network/NetworkGroup.cs#L110-L114
- `StorageGroup`, from DiskInfoToolkit's `DevicesChanged`, which is raised on its own monitoring thread when a disk is added or removed: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/677a3a56abde9adff5abdb42db3a7638c9137572/LibreHardwareMonitorLib/Hardware/Storage/StorageGroup.cs#L45-L63
- `HardwareNode.SensorAdded` and `SensorRemoved`, which add and remove sensor nodes whenever a hardware class calls `ActivateSensor` or `DeactivateSensor` during `Update()`, and the update runs on a `BackgroundWorker`: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/677a3a56abde9adff5abdb42db3a7638c9137572/LibreHardwareMonitor.Windows.Forms/UI/HardwareNode.cs#L105-L156

Coming back from sleep reconnects the network adapters, so that might explain the sleep reports here, and starting a Hyper-V VM in #2251 would do the same.  I haven't been able to reproduce it on demand though.

Skipping the bad index in `ClearNodesSize` hides one of the symptoms, but `CreateRowMap` and `NextNode` would still hit it.  I think the fix is to make every tree change happen on the UI thread, so `MainForm` and `HardwareNode` marshal their add and remove handlers with `BeginInvoke` when `InvokeRequired`.

closed 2026-10-12 by Blacktempel

depress77, 2026-11-02:
I'm seeing the BottomNode/CreateRowMap exception on LibreHardwareMonitor 0.9.6.0 (.NET 10 build, runtime 10.0.1, Aga.Controls 1.7.0.0). The stack goes through TreeNodeAdv.get_BottomNode() at line 264, TreeViewAdv.CreateRowMap() at line 900, and WinForms InvokeMarshaledCallbacks(), similar to #2442. A subsequent check showed CSV logging still working at one-second intervals; I don't yet have reliable reproduction steps or a confirmed trigger.

I noticed this issue was closed as completed on October 12, while the related UI-thread changes in #2452 were later reverted by #2572. I couldn't find those changes restored in the current MainForm/HardwareNode code. Could you clarify whether there is a replacement fix or a recommended build, and whether this issue should be reopened? What targeted diagnostics would help?

LaraSQP, 2026-11-04:
same problem

LibreHardwareMonitor 0.9.6

intermittent, but persistent and always when returning from sleep

```
System.ArgumentOutOfRangeException: Index was out of range. Must be non-negative and less than the size of the collection.
Parameter name: index
   at System.ThrowHelper.ThrowArgumentOutOfRangeException(ExceptionArgument argument, ExceptionResource resource)
   at System.Collections.Generic.List`1.get_Item(Int32 index)
   at Aga.Controls.Tree.TreeNodeAdv.get_NextNode() in D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeNodeAdv.cs:line 252
   at Aga.Controls.Tree.TreeNodeAdv.get_NextVisibleNode() in D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeNodeAdv.cs:line 281
   at Aga.Controls.Tree.TreeViewAdv.<get_VisibleNodes>d__261.MoveNext() in D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeViewAdv.Properties.cs:line 74
   at Aga.Controls.Tree.TreeViewAdv.CreateRowMap() in D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeViewAdv.cs:line 900
   at Aga.Controls.Tree.TreeViewAdv.UnsafeFullUpdate() in D:\a\LibreHardwareMonitor\LibreHardwareMonitor\Aga.Controls\Tree\TreeViewAdv.cs:line 694
   at System.Windows.Forms.Control.InvokeMarshaledCallbackHelper(Object obj)
   at System.Threading.ExecutionContext.RunInternal(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state, Boolean preserveSyncCtx)
   at System.Threading.ExecutionContext.Run(ExecutionContext executionContext, ContextCallback callback, Object state)
   at System.Windows.Forms.Control.InvokeMarshaledCallback(ThreadMethodEntry tme)
   at System.Windows.Forms.Control.InvokeMarshaledCallbacks()
```

clicking on 'continue' seems to work fine

but still