LibreHardwareMonitor/LibreHardwareMonitor #691: Wrong mutex used to synchronize access to Asus EC

argotronic, 2022-03-13:
Hi,

when checking your methods for accessing the ECs on modern Asus boards I found this as an attempt to synchronize access on SW level; from current master: 

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Motherboard/Lpc/EC/WindowsEmbeddedControllerIO.cs#L33

and according to:

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Ring0.cs#L96
the mutex used is `Global\\Access_ISABUS.HTP.Method`. While being the correct one for accessing SuperIO chips, for accessing those ECs, this is the wrong mutex to use!

Please change this to use `Global\\Access_EC` instead. This mutex was agreed by several monitoring tool creators, e.g. [HWiNFO](https://www.hwinfo.com/), [SIV](http://rh-software.com/), [AIDA64 ](https://www.aida64.com/) and [Argus Monitor](https://www.argusmonitor.com/) and using the ISABUS mutex means basically that there is NO synchronization at all at the moment on SW level. Especially for those ECs, this can lead to serious stability issues if LibreHardwareMonitor is accessing the ECs at the same time as any of the other tools.

Best regards
Udo

PhyxionNL, 2022-03-14:
Feel free to send a PR.

argotronic, 2022-03-14:
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/693

I hope those changes work (as I can only use a text editor in lack of a proper C# IDE).

closed 2022-03-15 by PhyxionNL