LibreHardwareMonitor/LibreHardwareMonitor #697: Alder Lake VCORE Readings Incorrect

seaweeduk, 2022-03-15:
Comparison with XTU, Libre, HWInfo64 and CPU-Z below. Libre is consistently showing much lower value for Vcore than everything else. Should show > 1.4V but never shows more than 1.3

[image]

seaweeduk, 2022-03-20:
As best as I can tell, the sensor which is being labelled as Vcore is actually VIN8 shown here in HWMonitor. I don't have any available sensor in Libre Hardware Monitor which matches the Vcore readings I should be seeing.

[image]

Motherboard is using Nuvoton NCT6796D

BlueTalon6, 2022-09-08:
Same here but much worse. Upgraded to an Asrock z690 and a 12700k and was shocked to see Vcore as high as 2.5v in LHM. Confirmed with other software, looks like it is showing roughly double actual voltage. 

[image]

Might be related to the old half voltage display issues in OHM (1238 and 1305)?

BlueTalon6, 2022-09-08:
I've noticed CPU core voltage is now listed under CPU (which appears to be accurate) so I guess the MB Vcore is inaccurate or requested/desired voltage