LibreHardwareMonitor/LibreHardwareMonitor #702: Support for NZXT RGB & Fan Controller

velocity7, 2022-04-22:
Product webpage here:
https://nzxt.com/product/rgb-and-fan-controller?sscid=41k6_nnsaz

NoFace33, 2022-11-05:
Is this still planned? I would love to not have to rely on NZXT Cam to manage my fans

CZonin, 2022-11-05:
It was implemented a couple months ago into LibreHardwareMonitor and now works with FanControl.

NoFace33, 2022-11-05:
I indeed saw the added support of NZXT Grid+ V3 a few months ago, but it never changed anything to my [AC-SDV02-A1](https://nzxt.com/en-CA/product/rgb-and-fan-controller) controller. Still no RPM reported to LHM and FanControl and no control availaible that speed.

CZonin, 2022-11-05:
Go into sensor settings and enable "Controller (various third parties)

NoFace33, 2022-11-05:
Unfortunately, it is already enabled and would not change the fact that even LHM does not see the data.

wutsch, 2022-11-07:
> What about RGB management? Is that something that could be managed from FanControl?

I use and recommend OpenRGB for RGB management.

nateKlaux, 2022-12-07:
Just came to second this post. It also isn't picking up any fans connected to the NZXT fan controller. Would really love for it to!

slhad, 2023-01-09:
I would also really love for the NZXT RGB & Fan Controller support to be implemented, as Grid + V3 is almost impossible to find now

LDannijs, 2023-02-15:
Ditto. Would like to ditch CAM for FanConrol but it's not showing up, even though it supposedly has support.

PierG2, 2023-02-16:
Any news regarding this thread? Each day CAM needs to be used is equal to a day of torture

lunchanddinner, 2023-07-22:
Same here, the NZXT fan controller still doesn't seem to show up even with "Controller (various third parties)" setting on. Any updates on this? Thanks!

repraze, 2023-08-05:
From the implementation, I don't think all NZXT controllers are supported.

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Controller/Nzxt/NzxtGroup.cs#L44C22-L44C28

Only `0x1711` seems to be integrated and that's for gridv3. My NZXT RGB & Fan Controller (https://nzxt.com/en-CA/product/rgb-and-fan-controller#) is on `0x2011`. I'm not sure if the same implementation would work since it has 3 fan ports and 6 rgb VS 6 fan ports of the gridv3.

sp00n, 2024-05-28:
I've already mentioned this in the other issue (#1305), posting it here as well since it's probably relevant.
[liquidctl](https://github.com/liquidctl/liquidctl) has support for a couple more NZXT controllers, e.g. the one with the id 0x2011, which is part of the H9 case.

Here's the relevant file:
https://github.com/liquidctl/liquidctl/blob/main/liquidctl/driver/smart_device.py

nateKlaux, 2024-05-30:
@sp00n after manually installing the liquidctl plugin I can't seem to get it to initialize and/or detect any sensors. Not sure if that means my NZXT controller isn't recognized of if it's another issue..

sp00n, 2024-05-30:
> @sp00n after manually installing the liquidctl plugin I can't seem to get it to initialize and/or detect any sensors. Not sure if that means my NZXT controller isn't recognized of if it's another issue..

I assume you mean the [plugin for FanControl](https://github.com/jmarucha/FanControl.Liquidctl)?
If so, yes, that plugin currently doesn't work for those NZXT controllers which return multiple fan entries (Fan 1 speed, Fan 2 speed, etc).
I created a modded version of the plugin that supports these controllers, but I probably violated a dozen of best practices for C# since I haven't really worked with it much before. It seems to work for my NZXT controller that was included in the H9 case though.
https://github.com/sp00n/FanControl.Liquidctl

nateKlaux, 2024-05-31:
@sp00n 

Ah yeah, I did mean for FanControl (didn't realize I was in the Libre repo...there are multiple issues discussing this). I'll give your modification a go and see if it works for me as well. Would be great to get rid of CAM.

lunchanddinner, 2024-06-01:
> @sp00n
> 
> Ah yeah, I did mean for FanControl (didn't realize I was in the Libre repo...there are multiple issues discussing this). I'll give your modification a go and see if it works for me as well. Would be great to get rid of CAM.

Let us know how it goes, and wow I haven't touched CAM in a while, have been using Signal to control my NZXT case fans

nateKlaux, 2024-06-03:
@sp00n your plug in worked for me!

lunchanddinner, 2024-06-03:
> I assume you mean the [plugin for FanControl](https://github.com/jmarucha/FanControl.Liquidctl)? If so, yes, that plugin currently doesn't work for those NZXT controllers which return multiple fan [...]

Just tested and it works on my NZXT controllers too!

sp00n, 2024-06-03:
Good to know.
I do receive an error message about once a day, where the log says that liquidctl.exe exited with a non-zero exit code. I'll probably have to add some debug output to see what's actually happening then.
Also, I'd really like someone with knowledge of C# to look over that code and correct the mistakes I probably made.

lunchanddinner, 2024-06-04:
> Good to know. I do receive an error message about once a day, where the log says that liquidctl.exe exited with a non-zero exit code. I'll probably have to add some debug output to see what's [...]

Yes I have that issue too, every restart of PC it will tell to refresh because liquid plugin could not be detected