LibreHardwareMonitor/LibreHardwareMonitor #708: Unable to manually change fan speed - amd 6800

Banapple01, 2022-04-23:
via FanControl https://github.com/Rem0o/FanControl.Releases a fan control project which uses this library. I am unable to change the fan speed of my gpu.

BenHartwich2005, 2022-04-23:
Also Aorus RX 5700xt is affected

SLCarlsson, 2022-04-23:
Also AMD RX 6700XT.

(Gigabyte B550i)

charmarkk, 2022-04-23:
> Also Aorus RX 5700xt is affected'

Sapphire PULSE 5700xt checking in.

BernardRobbins, 2022-04-25:
MSI Gaming Radeon RX 6600 XT isn't able to be controlled.

PhyxionNL, 2022-04-28:
If any of you have some coding experience, you can probably find out where it fails on around here (or a bit above):
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Gpu/AmdGpu.cs#L167-L215
I don't have an AMD GPU to test anymore since my RX 580 died ☹

johny-mnemonic, 2022-05-01:
@PhyxionNL I am not a coder, just a sysadmin, but if you are able to provide a debug build or debugging instructions I can give it a try on my RX 6800.

Misunderstood-Wookiee, 2022-05-11:
Also AMD 5600XT 
Unable to control the fan speed but detects temps just and fan % just fine, but fan control which uses this API is unable too control the speed.

I asked the Fan Control GitHub dude and they say it a LHM issue.
so here I am.

Misunderstood-Wookiee, 2022-05-11:
> If any of you have some coding experience, you can probably find out where it fails on around here (or a bit above): [...]

I can give it a crack...

The thing is I don't know what they have changed, the AMD Driver can control the fan but obviously, that is like secret mcnugget sauce so I have no way to see what inputs they are giving to the GPU I suspect at some point they changed the programming in the drivers and which breaks your original code.

PhyxionNL, 2022-05-11:
Feel free to give it a try 🙂 As long as you have some coding experience you should at least be able to find out what happens (or doesn't).

I don't think they changed the APIs, it's still ADL (see https://gpuopen-librariesandsdks.github.io/adl/group__OVERDRIVE5API.html), but it might be that these cards needs to use a newer version such as ADL2_Overdrive6_FanSpeed_Set or ADL2_OverdriveN_FanControl_Set.

Misunderstood-Wookiee, 2022-05-11:
If know where to start I can probably build the source with newer values
for testing will see what happens if I find a fix I'll fork and PR

Rem0o, 2022-06-14:
Made an attempt at some point but lacked support.
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/386

katiekloss, 2022-07-17:
Poked at this for a few hours, and I think the issue is that Overdrive 8+ doesn't support anything other than fan curves. On my RX 6900 (Overdrive 8), I get ADL_ERR_NOT_SUPPORTED from every API function that can set a specific fan speed.

[There's a sample program available](https://github.com/GPUOpen-LibrariesAndSDKs/display-library/blob/79cdb8e22b8650c861b390d5a9895032492ce8c5/Sample/Overdrive8/Overdrive8.cpp#L2154) that seems to accomplish it by setting all of the fan curve outputs to the same value, which seems viable (though inefficient).