LibreHardwareMonitor/LibreHardwareMonitor #709: Add support for Asus B450-F Gaming.

KonouChan, 2022-04-23:
Via https://github.com/Rem0o/FanControl.Releases. For some reason, only Fan Control 4 (Fan Chassis 3) out of 3 other fans connected (including cpu) are controllable, but sensors are detectable through HWmonitor and GPU (Nvidia 1660s gigabyte) is detected and controllable.

KonouChan, 2022-04-23:
After reading through a crap ton of stuff based on my motherboard, it is using ITE8665E. From what I have gained, there's not a lot about this chip and I guess it requires some poking to find out which address controls what for this specific motherboard since it differs? (Which is why only Fan Chassis 3 is working but the rest is not?). Are there any tools that can detect which address controls what?

Svek, 2022-06-17:
A quick update... I've added some code to get into this issue further.
Some triage and a bit more testing might be needed. (https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/758)

michlbro, 2023-08-14:
Just want to bump this. Been about a year and a month and hoped for a bit of progress, but I don't think it has happened yet.

Svek, 2023-08-14:
> 

would you be open to running a specially compiled version that will dump out all the data, so we can map it out?

michlbro, 2023-08-15:
Sure!

michlbro, 2024-12-01:
I just want to bump that #1438 has not yet fixed the issue with Asus B450-F. Honestly, I feel like any Asus related motherboards are a lost cause and I'm just going to switch to a different motherboard brand.