LibreHardwareMonitor/LibreHardwareMonitor #712: Asus P7P55D-E CPU Fan (PWM) can not be controlled.

itplayer-de, 2022-04-24:
Hello together,

I know the mainboard is ancient, but I still use it. The control of the System Fan works without problems, only the CPU Fan can't be assigned and even if you do this manually, the speed can't be controlled. Others have probably also had problems with this: http://www.bugtrack.almico.com/view.php?id=1528
Maybe someone has the time to take a look at it. If you need further information, please contact me at any time.

Mainboard: Asus P7P55D-E
CPU: Intel i7-860
"Super I/O": Windbon/Nuvoton W83667HG
Os: Windows 10 21H2

Thanks and all the best
Chris

Svek, 2022-06-17:
Would you help detail a bit further, if the RPM is being displayed or not?

itplayer-de, 2022-06-17:
Sorry, yep RPM is displayed.

Tobim6, 2024-06-30:
I have the ASUS P7P55D-E PRO and have the exact same issue

Tobim6, 2024-08-15:
I would like this to get some attention please.

Tobim6, 2025-03-16:
I don't think we are getting anything from this...