LibreHardwareMonitor/LibreHardwareMonitor #714: [FanControl] cannot correctly control fans on Asus TUF x570 plus

ErisaFirehawk, 2022-04-25:
Completely new here, but I seem to be having a big issue controlling the fans speeds. When controlling them, rather it is manual or via a graph, the fan never goes above about 650 RPM. the moment i disable the control, it speeds up if that's where it is in the bios curve. Also, I have the control for my water pump connected to the main CPU header, can that wont respond to any commands from the software. I have confirmed that the fan and sensor is matched correctly by unplugging the pump control and the RPM drops to 0.

the only one that seems to work correctly is the chipset fan on the motherboard.

Windows 11
Asus TUF x570 plus motherboard
Ryzen 3600

I have tried making sure HWmonitor is open first, as well as tried opening cpu-z first, and i get the same thing each time.
bios control
[image]

FanControl
[image]

quangkieu, 2022-05-02:
this is Asus AI Suite fan control "feature". If you are using quiet mode, it would cap the low and high speed of the fan. And if you use PWM mode in Bios Fan Control then it would cap the low speed so you could not shut it down.

ErisaFirehawk, 2022-05-02:
> this is Asus AI Suite fan control "feature". If you are using quiet mode, it would cap the low and high speed of the fan. And if you use PWM mode in Bios Fan Control then it would cap the low speed so you could not shut it down.

yeah, i just uninstalled the AI Suite and most everythign is working like it should, though i still cant control the main CPU fan header (my pump pulls a pwm signal from it to run its speed)

Svek, 2022-06-14:
Just to clarify, when you say "can't control the main CPU fan header". Does that mean the sensor is showing the correct RPM values, but you are only not able to control the speed?

Svek, 2022-06-17:
When you set the power down to 0%, 50%, 100%, is the fan (in your case, pump) physically changing it's speed?

What I am trying to understand, is if it just the RPM value is wrong, but the pump is actually going at 100%.

i.e. The UI shows 650RPM, but in fact, it's running at 4000RPM. Therefore, the controls for the speed is working, and it's just a "cosmetic" issue at hand.

ErisaFirehawk, 2022-06-17:
the RPM value seem right, but the slider doesnt do anythign to the pump speed.

Svek, 2022-06-17:
Ok. That helps to clarify the issue a bit further.
So, even when you set it to 0% it stays constant at ~650RPM, correct? Nothing changes, the pump is always the constant 1 speed?

Svek, 2022-06-17:
One more thing, could you help confirm that your BIOS currently has which setting for your FAN1? I believe the three options are `Auto`, `PWM` or `DC`?