LibreHardwareMonitor/LibreHardwareMonitor #731: Gigabyte X570 Aorus Master - ITE IT8792E - No controls

Gucioo, 2022-05-02:
This motherboard has 2 IO chips:
- for 1st one sensors and controls are present
- for 2nd there are only sensors, no controls: Fan 4, 5 & 6.

 It's a ITE IT8792E. How support for this could be added?

[image]

Dirkxke, 2022-05-08:
The X570 Aorus Pro uses the same controller for the same fans.
2 birds 1 stone?

kybdi, 2022-06-12:
I am having the same issue as described, where the second group of sensors is being ignored by fan controls.

[image]

Aorus X570 Extreme is my motherboard, looks like the value is showing as it8792e/it8795e

I am unsure if adding just one will fix this issue, so if possible (assuming they are closely related, could support for t8792e AND it8795e be added?

mrehkopf, 2022-07-11:
Same issue here with X570S AORUS MASTER, I can read and control the fans on the IT8689E.

But the Fan Controls on the IT8695E act as if they are read only.
The RPM control setting seems to reflect the current setting on the controller (66%) but I cannot change it.
When I stop the fan by hand the Fan RPM reading (Fan #3) changes accordingly.

[image]

slhad, 2022-07-13:
I got a X570 Aorus Pro and I have also no control of the 3 fans attached to ITE IT8792E/IT8795E like everybody here

slhad, 2022-07-13:
Seems like there is an old issue with new comments on it referencing the problem #251
Apparently ITE IT8792E/IT8795 controllers rewrite the old control value of the bios just after the user change it

Feel like it will be quicker to change the motherboard than have some sort of update of bios/drivers that fix this weird behavior

severgun, 2022-09-12:
> Apparently ITE IT8792E/IT8795 controllers rewrite the old control value of the bios just after the user change it
> Feel like it will be quicker to change the motherboard than have some sort of update of bios/drivers that fix this weird behavior

I have Gigabyte Z690 PRO with 8695 that behave same way. Read PWM % but not set. Gigabyte SIV software can control it, so I would say that it is not BIOS issue.
Seems like whole https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Motherboard/Lpc/IT87XX.cs just a mess. Should be error with registers or values somewhere.
To be fair I can't find datasheets for most of ITE products.

PhyxionNL, 2023-03-15:
See #731.

closed 2023-03-15 by PhyxionNL

BobC0728, 2023-05-17:
Any hope for this?  Also an Aorus Master X570S MB.  Seems like it has been an issue for two years plus.