LibreHardwareMonitor/LibreHardwareMonitor #737: Incorrect reading from ASUS TUF B450-PRO GAMING (ITE IT8665E)

CovenantSquad, 2022-05-07:
Hi, 
first of all, thanks for all of your work in this project.
LibreHardwareMonitor is reading wrong temperature and fan speed values from my B450-PRO (ITE IT8665E); in fact FanControl, which uses this library to read temperatures and set fan speeds, is not able change any of the attached fans' speed.
It seems like both LibreHardwareMonitor and FanControl work correctly when CPUID HWMonitor is (or has been) running.
![Screenshot (36)](https://user-images.githubusercontent.com/84512066/167247038-4b4a5bf4-a29b-4275-bcdf-5a18e7fb0922.png
[image]
)

Svek, 2022-06-17:
I'm looking into all these ASUS board related issues and particularly this statement you are making
> [works] correctly when CPUID HWMonitor is (or has been) running.

I saw a few other threads about this in other parts of the internet, and I have a hunch this has to do with permissions and potentially some outdated or bugged code from OpenHardwareMonitor (where a good portion of the underlying code that LHW relys on to "just work").

Svek, 2022-06-17:
A quick update... I've added some code to get into this issue further.
Some triage and a bit more testing might be needed.

https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/758

CovenantSquad, 2022-06-17:
That's awesome, Thanks!

PR #1443 (2024-09-20): Fix: Fix fan and temp data for IT8665E chip motherboards, fixes #737 and more
This adds the same fix from #1438 to IT8665E chips, fixing a bug where speeds and other sensor data is incorrectly reported/not shown at all. Fixes #737 and a few other issues.
commit 993f9d4b (2024-09-20): Fix: Fix fan and temp data for IT8665E chip motherboards, fixes #737 and more (#1443)

* Fix fan and temp data for IT8665E chip motherboards

* updated IT8665E to 4 Fans

---------
closed 2024-09-20 by PhyxionNL