LibreHardwareMonitor/LibreHardwareMonitor #744: Z590 Aorus Master

Samm223, 2022-05-18:
my motherboard is missing a few fans that will not pair with fans that are running on my motherboard. looking to see if there is a fgix in the future for this?

Svek, 2022-06-17:
It would be helpful to see if HWiNFO shows these fans listed under a separate EC.

This was the case for my ASUS board, but I don't have any idea about Gigabyte's way of handling this.

Samm223, 2022-07-10:
I Downloaded HWinfo, where do i look to find my fans?

Samm223, 2022-07-10:
[image]

i found how to get a picture of them

Samm223, 2022-07-10:
[image]

her is all the fans on HWinfo, let me know what else you need

tprochazka, 2023-10-02:
It looks like it is an issue with almost all Gigabyte motherboards, there is always a second chip that is just partially supported. I can see RPM, and I can even control it, but It can't read the current Fan speed control on Z790 master
[image]

tprochazka, 2023-10-09:
Actually, the one with "-" instead of real values is possible to control, and the second one which show all the values not. Is there any workaround on how to do that?