LibreHardwareMonitor/LibreHardwareMonitor #749: Microsoft .NET Framework ERROR on Startup W11

8Christian, 2022-05-28:
Informationen über das Aufrufen von JIT-Debuggen
anstelle dieses Dialogfelds finden Sie am Ende dieser Meldung.

************** Ausnahmetext **************
System.InvalidOperationException: Die Sammlung wurde geändert. Der Enumerationsvorgang kann möglicherweise nicht ausgeführt werden.
   bei System.ThrowHelper.ThrowInvalidOperationException(ExceptionResource resource)
   bei System.Collections.Generic.List`1.Enumerator.MoveNextRare()
   bei LibreHardwareMonitor.Wmi.WmiProvider.Update()
   bei LibreHardwareMonitor.UI.MainForm.Timer_Tick(Object sender, EventArgs e)
   bei System.Windows.Forms.Timer.OnTick(EventArgs e)
   bei System.Windows.Forms.Timer.TimerNativeWindow.WndProc(Message& m)
   bei System.Windows.Forms.NativeWindow.Callback(IntPtr hWnd, Int32 msg, IntPtr wparam, IntPtr lparam)

************** Geladene Assemblys **************
mscorlib
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.4515.0 built by: NET48REL1LAST_C.
    CodeBase: file:///C:/Windows/Microsoft.NET/Framework64/v4.0.30319/mscorlib.dll.
----------------------------------------
LibreHardwareMonitor
    Assembly-Version: 0.9.0.0.
    Win32-Version: 0.9.0.0.
    CodeBase: file:///D:/Tools/LibreHardwareMonitor/LibreHardwareMonitor.exe.
----------------------------------------
System.Windows.Forms
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.4515.0 built by: NET48REL1LAST_C.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms/v4.0_4.0.0.0__b77a5c561934e089/System.Windows.Forms.dll.
----------------------------------------
System
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.4488.0 built by: NET48REL1LAST_C.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll.
----------------------------------------
System.Drawing
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.4395.0 built by: NET48REL1LAST_B.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Drawing/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Drawing.dll.
----------------------------------------
Aga.Controls
    Assembly-Version: 1.7.0.0.
    Win32-Version: 1.7.0.0.
    CodeBase: file:///D:/Tools/LibreHardwareMonitor/Aga.Controls.DLL.
----------------------------------------
LibreHardwareMonitorLib
    Assembly-Version: 0.9.0.0.
    Win32-Version: 0.9.0.0.
    CodeBase: file:///D:/Tools/LibreHardwareMonitor/LibreHardwareMonitorLib.DLL.
----------------------------------------
Microsoft.Win32.TaskScheduler
    Assembly-Version: 2.10.1.0.
    Win32-Version: 2.10.1.0.
    CodeBase: file:///D:/Tools/LibreHardwareMonitor/Microsoft.Win32.TaskScheduler.DLL.
----------------------------------------
System.Xml
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.4161.0 built by: NET48REL1.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Xml/v4.0_4.0.0.0__b77a5c561934e089/System.Xml.dll.
----------------------------------------
System.Core
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.4515.0 built by: NET48REL1LAST_C.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Core/v4.0_4.0.0.0__b77a5c561934e089/System.Core.dll.
----------------------------------------
System.Configuration
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.4341.0 built by: NET48REL1LAST_C.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.dll.
----------------------------------------
System.Management
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.4435.0 built by: NET48REL1LAST_C.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Management/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Management.dll.
----------------------------------------
OxyPlot
    Assembly-Version: 2.1.0.0.
    Win32-Version: 2.1.0.0.
    CodeBase: file:///D:/Tools/LibreHardwareMonitor/OxyPlot.DLL.
----------------------------------------
OxyPlot.WindowsForms
    Assembly-Version: 2.1.0.0.
    Win32-Version: 2.1.0.0.
    CodeBase: file:///D:/Tools/LibreHardwareMonitor/OxyPlot.WindowsForms.DLL.
----------------------------------------
System.Configuration.Install
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.4161.0 built by: NET48REL1.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Configuration.Install/v4.0_4.0.0.0__b03f5f7f11d50a3a/System.Configuration.Install.dll.
----------------------------------------
hsfopcvd
    Assembly-Version: 0.0.0.0.
    Win32-Version: 4.8.4488.0 built by: NET48REL1LAST_C.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System/v4.0_4.0.0.0__b77a5c561934e089/System.dll.
----------------------------------------
mscorlib.resources
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.4161.0 built by: NET48REL1.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/mscorlib.resources/v4.0_4.0.0.0_de_b77a5c561934e089/mscorlib.resources.dll.
----------------------------------------
System.Windows.Forms.resources
    Assembly-Version: 4.0.0.0.
    Win32-Version: 4.8.4161.0 built by: NET48REL1.
    CodeBase: file:///C:/Windows/Microsoft.Net/assembly/GAC_MSIL/System.Windows.Forms.resources/v4.0_4.0.0.0_de_b77a5c561934e089/System.Windows.Forms.resources.dll.
----------------------------------------

************** JIT-Debuggen **************
Um das JIT-Debuggen (Just-In-Time) zu aktivieren, muss in der
Konfigurationsdatei der Anwendung oder des Computers
(machine.config) der jitDebugging-Wert im Abschnitt system.windows.forms festgelegt werden.
Die Anwendung muss mit aktiviertem Debuggen kompiliert werden.

Zum Beispiel:

<configuration>
    <system.windows.forms jitDebugging="true" />
</configuration>

Wenn das JIT-Debuggen aktiviert ist, werden alle nicht behandelten
Ausnahmen an den JIT-Debugger gesendet, der auf dem
Computer registriert ist, und nicht in diesem Dialogfeld behandelt.

Kommprog, 2022-06-05:
Windows 7 also has the same bug/.net error not only after reboot, but after waking up from sleeping mode.

closed 2022-06-27 by PhyxionNL

riQQ, 2022-06-27:
Are #385 and #460 also fixed by 54a407d?

PhyxionNL, 2022-06-28:
> Are #385 and #460 also fixed by [54a407d](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/commit/54a407dc50149e255209e4642f7bbcd828fc36fa)?

No, different errors.

commit 54a407dc (2022-06-27): Fixes #749