LibreHardwareMonitor/LibreHardwareMonitor #755: ASUS - Z10PE-D16 WS   No fan sensors found :(

JamesSauter, 2022-06-13:
Seems Asus mb's are an issue. Fan Control isn't finding any fan sensors to control, but I can see all the fan speed in this BIOS
Looks like it found all 5 running fans, it just can't control them. :(

[image]

Svek, 2022-06-14:
If you could get another application like HWiNFO (https://www.hwinfo.com) and pull a screenshot of that, we can take a look if the  chip is defined. I don't have this board, so to support it, we may need your help to pull additional information about your hardware so that the code can be updated to support it.

JamesSauter, 2022-06-14:
Hope this helps too :)

[image]

Svek, 2022-06-17:
Quite unusual, it might just be something related to FanControl. More specifically, the version of the `LibreHardwareMonitorLib.dll` that it is using, otherwise it might not be initializing properly.

Did you attempt to use LibreHardwareMonitor? https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/releases/download/v0.9.0/LibreHardwareMonitor-net472.zip

Would you mind trying this application and see if it is showing?

Assuming that the fans do show up here, then it's rather easy to fix.

Svek, 2022-06-17:
A quick update... I've added some code to get into this issue further.
Some triage and a bit more testing might be needed. (https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/pull/758)

JamesSauter, 2022-06-17:
This is all new to me...  I hope this helps. :)
[LibreHardwareMonitor.Report-1.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/8927390/LibreHardwareMonitor.Report-1.txt)

Rem0o, 2022-06-19:
@JamesSauter 

As @Svek said, do you get readings from the standalone LHM app?

JamesSauter, 2022-06-19:
> @JamesSauter
> 
> As @Svek said, do you get readings from the standalone LHM app?

Not sure...  Looks like is sees the sensors now.  I put them all into manual mode and turned them all off, but nothing changed in the case that I could hear.  Then again, there are 2 Xeon fans, 3 GPU fans, 3 Chassis fans and the 1500w PS fan. Maybe it's just going to be loud no matter what. 
[image]

Svek, 2022-06-22:
At the moment, it would be helpful if you could try downloading LibreHardwareMonitor (LHM) and see what the fan values are on the application.

If it only detects Fan#1 through Fan#5 with no RPM readings, then it confirms the issue is with LHM and not FanControl.

We are trying to narrow down the root cause.

Svek, 2022-07-07:
@JamesSauter did you have a chance to download LHM and test if it shows any values?

VirusDr, 2022-07-17:
have the same problem with a ASUS Pro WS WRX80E-SAGE SE WIFI, and for me in LHM app the fan shows 0 rpm.
One thing that I notice is that (for me) the fans are not named Fan #1, #2... 
CPU_FAN
CPU_OPT
CHA_FAN1
CHA_FAN2
CHA_FAN3
CHA_FAN4
CHA_FAN5
CHA_FAN6
SOC_FAN

HWiNFO64 can read the RPM on the fans and shows the "right" names.

AnnaKutou, 2024-08-15:
> @JamesSauter did you have a chance to download LHM and test if it shows any values?

I have same issues on ASUS Z10-PA-D8(same series with Z10PE-D16), here is screenshot on LHM
[image]

AnnaKutou, 2024-08-15:
@Svek Also, Voltages and Temperatures seems not correct. Is some wrong with NCT6779D? 
HWiNFO64 shows there is both sensor_IC named NCT6779D and NCT7904D, the fan's RPMs in NCT7904D are correctly read.
Here is screenshot on HWiNFO64
[image]