LibreHardwareMonitor/LibreHardwareMonitor #760: ASUS PRIME B450-PLUS: fan control doesnt work and fan sensors show incorrect RPM.

KAMKillJoy, 2022-06-19:
[image]
[image]
I see a lot issues about same problem on wide variety of Asus motherboards so here I am.
Just one more thing I discovered: usually i able to control one out of four my fan controllers, but this number randomly increases if "asus ai suite" (official asus utility for many thing and especially fan control) is working. This behavior is very inconsistent and controllable fans changing every time.

Rem0o, 2022-06-19:
Keep in mind that Asus ai suite and LHM  ***WILL*** conflict with each other if can both monitor/control fans.

KAMKillJoy, 2022-06-20:
> Keep in mind that Asus ai suite and LHM _**WILL**_ conflict with each other if can both monitor/control fans.

I know and not gonna use asus ai suit. It has already been uninstalled.

KAMKillJoy, 2022-07-08:
> Which version of LHM are you using?

0.9.0

SimonZerafa, 2023-04-02:
Hi,

Just to confirm that I am seeing the same issue with the latest build of LHM on ASUS B450-PLUS with UEFI Firmware 3211 (08/10/2021). The controller is reported as the ITE IT866SE in HWINFO. CPU fan speed is not detected at all, however CPU Temperature is reported and this appears to be accurate,

The CPU fan is connected to CPU_FAN header with a 4 wire connector and selecting either PWM or AUTO has no effect on how LHM fails to detect the fan control or RPM.

PR #1438 (2024-08-22): Bugfix: Fan and Temp data missing #780
This is a fix for #780 IT8655E which shows incorrect/missing information for fans and temps.  Also fixes #614.

I have tested this on my ASUS Prime B350M-A, through a number of restarts, sleeps and hibernates, and I no longer see the incorrect fan speeds, incorrect temperatures or missing temperatures.

Inspiration came from [Linux it87](https://github.com/frankcrawford/it87/blob/14b1de2fb49db4e73b58177bc43daa71ceb8e92f/it87.c#L3870) where the interaction with the IT8655E does a set_bank before every interaction with a port.

In theory this fix can be app [... cut]