LibreHardwareMonitor/LibreHardwareMonitor #776: Cannot access CPU temperature on intel 12th gen processor.

liblaf, 2022-07-10:
Cannot access CPU temperature on intel 12th gen processor.

frankfoxy, 2022-07-13:
Not work for me on 12700H
[image]

DevTechProfile, 2022-07-13:
Please post a CPU-Z screenshot. This is Alder Lake-H, there is a problem with a missing model ID.

DevTechProfile, 2022-07-13:
Created PR  #779.

liblaf, 2022-07-14:
Not work for me on i7-1260P

[image]

[image]

DevTechProfile, 2022-07-15:
> Not work for me on i7-1260P

I've added model ID 9A to the switch-case-block, see link to PR above.

PhyxionNL, 2022-07-15:
Done 👍

closed 2022-07-15 by PhyxionNL