LibreHardwareMonitor/LibreHardwareMonitor #797: D3D 3D sensor is always 0

Jamesits, 2022-08-04:
Expected:
"D3D 3D" sensor should have some value when a DirectX 3D application is rendering something.

Actual:
It is always zero both in the GUI and from the data returned by LibreHardwareMonitorLib. See the screenshot below:
[image]

System information:
- Hardware: i7-11700F/RTX3060 (VBIOS 94.04.71.00.87)
- Nvidia driver version: 516.59
- LibreHardwareMonitor version: tried 0.9.0, 0.9.1-pre136, 0.9.1-pre140, master ([ae3bc27](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/commit/ae3bc2771978f458ac97efe8fc16ecab11f14fbb)), the result is the same

Levminer, 2023-03-15:
I can also confirm this issue exists.

```
Name: NVIDIA GeForce GTX 1660
Index: 0
Driver Version: 527.56
Driver Branch: r527_32-11
DeviceID: 0x218410DE
SubSystemID: 0x86D01043
RevisionID: 0xA1
ExtDeviceID: 0x2184

PhyxionNL, 2023-07-07:
This is because the RunningTime from 3D (and various others) is always 0. Only Copy and Video Decode seem to be working. Using performance counters https://github.com/winsiderss/systeminformer/blob/master/plugins/ExtendedTools/counters.c#L421 works OK but needs quite some changes. Up for grabs.

Levminer, 2024-02-19:
Hi @PhyxionNL!

I made a POC (https://github.com/Levminer/cores/blob/dev/platforms/windows/lib/GPUUsage.cs)

It could use some improvement, like checking if a performance counter still exists, and not using try/catch for that.

I'd love to make a PR, just point me to the code where I should add it.