LibreHardwareMonitor/LibreHardwareMonitor #806: Speeds in tray

pasha-zzz, 2022-08-29:
Hi. When any of speeds in B/KB/MB/s enabled to show in tray - it shows cutted center of value. This is totally useless.
F.e. network download/upload speeds, read/write rates of disks, PCIex throughputs.

pasha-zzz, 2022-09-08:
What kind of help you want?

HypnoB, 2023-01-24:
Kinda related to this issue, would it be possible to add an option to force a certain unit for network speed etc, instead of having it be dynamic? 

And maybe the option to select how many decimal places to display?

oefi, 2025-02-17:
This is how the Ethernet section 'Throughput' icons look like. It is very hard to guess what it is showing. The above mentioned options would be great. Thanks!

[image]