LibreHardwareMonitor/LibreHardwareMonitor #809: Is there any way to output JSON data on the command line or just start the server?

LRF520, 2022-09-01:
Is there any way to output JSON data on the command line or just start the server? Because the GUI interface consumes too much CPU, it always occupies 13% of the CPU. Most of the time, only JSON data can be used without starting the GUI.

[image]

JohnTravolski, 2022-09-03:
I use python to grab it

LRF520, 2022-09-08:
> I use python to grab it
> 
> ```
> from urllib.request import urlopen
> import json
> mylink = f'http://{ip}:8085/data.json'
> pagesource = urlopen(mylink).read().decode('utf-8')
> alldata = json.loads(pagesource)
> ```

It still needs to start the whole GUI before it can grab JSON data. Is there a way to start only the server now?

JohnTravolski, 2022-09-09:
I am not sure. I have a Windows task to run it at Windows startup and I have it set to be minimized to tray by default.

[image]

LRF520, 2022-09-09:
> I am not sure. I have a Windows task to run it at Windows startup and I have it set to be minimized to tray by default.
> 
> [image]

It always occupies 13% of the CPU,reducing CPU consumption is the problem I want to solve at present.

closed 2022-09-14 by LRF520