LibreHardwareMonitor/LibreHardwareMonitor #811: 12th generation Intel CPU can not get temperature

MuggleTan, 2022-09-10:
Hello, I use the 12th Intel i7-12700 CPU and Windows11, but the software can't get the temperature of the CPU.
[image]

severgun, 2022-09-10:
Maybe H related
[image]

iophobia, 2022-09-10:
No temps here on an i5-1240P either. Win 10 Pro w/ LHM 0.9.0

PhyxionNL, 2022-09-14:
Can you attach a report? File > Save Report.

iophobia, 2022-09-14:
Sure, here you go. [LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/9564634/LibreHardwareMonitor.Report.txt)

closed 2022-09-14 by PhyxionNL