LibreHardwareMonitor/LibreHardwareMonitor #813: Cannot access CPU temperature on 12th Gen Intel Core i7-12700F

HackHarry, 2022-09-11:
[image]

BlueTalon6, 2022-09-13:
If you scroll down, in temperatures is there "CPU package" under "CPU Core #12"?

HackHarry, 2022-09-13:
Yes, it is.
[image]

BlueTalon6, 2022-09-13:
Sorry, I thought you were looking for the sensor, didn't notice your values were blank. 
I notice there is no chip listed under your motherboard and no sensors listed so I suspect it is not recognised or doesn't provide the necessary readouts

PhyxionNL, 2022-09-14:
Can you attach a report? File > Save Report.

sebastian-dev, 2022-09-14:
https://github.com/torvalds/linux/blob/master/arch/x86/include/asm/intel-family.h
#define INTEL_FAM6_ALDERLAKE		0x97
#define INTEL_FAM6_ALDERLAKE_L 0x9A
#define INTEL_FAM6_ALDERLAKE_N		0xBE

#define INTEL_FAM6_RAPTORLAKE		0xB7
#define INTEL_FAM6_RAPTORLAKE_P		0xBA
#define INTEL_FAM6_RAPTORLAKE_S		0xBF

PhyxionNL, 2022-09-14:
Thanks @sebastian-dev, PR #815.

PR #815 (2022-09-14): Add RaptorLake and one more Alder Lake
Fixes #813, #811
closed 2022-09-14 by PhyxionNL