LibreHardwareMonitor/LibreHardwareMonitor #826: Unload driver on exit

Symbai, 2022-09-20:
No idea why this library needs a driver at all but the file/driver/sys file remains locked until computer restart. Please make the driver unload when application exit.

PhyxionNL, 2022-09-20:
This is already done.
https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Ring0.cs#L381

closed 2022-09-20 by PhyxionNL

Symbai, 2022-09-20:
No it's not. All the code you linked to does is trying to delete the file. But obviously this doesn't work because the driver is loaded and the file remains locked. You have to UNLOAD the driver first before you can delete the file. And this isn't done or it simply doesn't work.

PhyxionNL, 2022-09-21:
It does unload it, https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/blob/master/LibreHardwareMonitorLib/Hardware/Ring0.cs#L357
I've tried it on various computers and it always unloads correctly. If it doesn't for you, feel free to find out why and submit a PR to fix.

Symbai, 2022-09-21:
You can reproduce it when you kill the application in VS (for example ending debugging). Then open it again and then close it properly. Then you're not able to delete the driver, because its still loaded from first session. Shouldn't it get unloaded from second session? I mean it re-uses the driver right?

PhyxionNL, 2022-09-21:
The ref count will obviously not decrease then, which will prevent the driver from unloading. This is a generic driver that's also used by other applications so you cannot unload it as it would break them. If you don't want this to happen, don't kill the application but close it properly.

Symbai, 2022-09-21:
You cannot prevent your application from getting closed unexpectedly due to external factors. For what exactly does this library needs a driver and can I use it somehow without the driver?

PhyxionNL, 2022-09-21:
The driver is needed for a lot of things, you can check the source. A heartbeat isn't possible, the driver is available as-is and would need to be modified and then resigned with a certificate. So unless you want to make the necessary changes and purchase an ev code signing certificate, just close it properly or keep the driver loaded 🙂 
That said, if you really want to, you can force to unload the driver yourself, but LHM won't do that for the reason mentioned above.