LibreHardwareMonitor/LibreHardwareMonitor #833: feat: add Overdrive 8 (AMD GPU fan control)

antt0n, 2022-10-02:
Hello to all,

I would like to make a general issue about the implementation of Overdrive 8. The subject comes back to many other issues that raise the impossibility to act (especially) on the control of the fans of the new AMD graphics cards.
Overdrive 8 implements a new fan curve system, which makes the old fan control inoperable.
I can try to look into it, I will do some tests on my side, but I don't think I have the time and the skills to do it.
I would have to rework the logic behind the fan control to implement this curve system, and I would have to think about whether to use this curve functionality proposed by OD8 (based on a temperature and a speed, in the maximum of 4 points), or to implement the system as it was presented before, simply by giving the graphic card a speed for the fans.

xhemp, 2023-01-02:
Found this tool which looks like you can change the fans curves ( I have a 7900 XTX, tested and looks like it works), project looks dead (dev hasn't logged in on guru3d.com since February 2021):

https://forums.guru3d.com/threads/overdriventool-tool-for-amd-gpus.416116/

Application is here:

https://www.dropbox.com/s/gofwpz586qrpvsr/OverdriveNTool%200.2.9.7z?dl=1 

Source code for that application is here, maybe it helps?

https://www.dropbox.com/sh/40lhboermnnscxf/AAD6byI7baclQXupYzv2GuyWa?dl=0

Rem0o, 2023-02-01:
Found this new thing https://github.com/GPUOpen-LibrariesAndSDKs/ADLX

Should support RDNA3 cards.  There is a C# sample, but it is only configured with Swig to generate a couples of modules, not all of them, especially the interesting ones (GPUTuning, GPUManualFanTuning.).   I don't know much how to edit the "i" files to get everything generated, including the headers wasn't enough. There seem to be some gibberish that needs to be added in as a template to tell which functions to export. 

I will fork this and try and make a working .NET library out of it.

Rem0o, 2023-02-01:
Forked said repo:

https://github.com/Rem0o/ADLX/tree/main/Samples/csharp

Made a c# project out of it, with a test console app. If somebody with an RDNA3 card with the latest driver installed could test it out, would be appreciated.

xhemp, 2023-02-01:
Just ran the sample:

Setting fan speed to 1000 rpm...
Temp is 51
Fan speed is 0

D:\git\ADLX\Samples\csharp\ADLXWrapper.TestConsole\bin\Release\net48\ADLXWrapper.TestConsole.exe (process 13092) exited with code 0.

antt0n, 2023-02-01:
Hello,

I just did the test under Windows 11 x64 with the latest drivers with an RX 6700 (RDNA 2), and I have this error message.
`ADLX result (ADLX_FAIL): Couldn't initialize ADLX`

antt0n, 2023-02-01:
I took the testing a little further, and this is what I got when I tested the following test project: https://gpuopen.com/manuals/ADLX/adlx-cpp__manual_fan_tuning/#code.
[image]

antt0n, 2023-02-01:
> @xhemp @antt0n
> 
> The fact it is even working a bit is very promising!
> 
> There seems to be 2 ways to set fan speed. SetTargetFanSpeed which may or may not be supported, and the state list. For the state list, I will simply set the same speed for all temperatures.
> 
> Did an update on the repo. Test again and show the result.

```
Set target fan speed is supported: False
States error index is -1
Temp is 35
Fan speed is 2144
```

xhemp, 2023-02-01:
Got the same:

Set target fan speed is supported: False
States error index is -1
Temp is 42
Fan speed is 0

antt0n, 2023-02-01:
> Got the same:
> 
> Set target fan speed is supported: False States error index is -1 Temp is 42 Fan speed is 0

I think that the application of the parameters (Fan speed) did not take place because the AMD parameters allow zero RPM, and it must be managed directly by the drivers of when to activate the fans.
(this is a guess, I have no idea if it comes from there)

xhemp, 2023-02-01:
I was playing with the tool I posted above (Overdrive) and setting 0 RPM I was able to change the fans speeds and temperatures by zone (P0 to P5), it worked so it's definitely possible.
[image]

Fans were going nuts at 78%

Rem0o, 2023-02-01:
@antt0n  soooo from your screenshot it seems to be working? All state point at 50% for all temps.

Rem0o, 2023-02-01:
@antt0n Disable zero rpm first in adrenaline, then test my latest commit, which should turn it back on....

Rem0o, 2023-02-01:
Also just realized, try and disable the "Contrôle avancé" (advanced control) toggle then run the program again, see if target fan speed works then.

antt0n, 2023-02-01:
[image]
[image]
[image]

antt0n, 2023-02-01:
[image]
[image]
I have tested all possible configurations (Zero RPM on and off, with Advanced Control on and off each time), and the settings always come back as shown on the screen.
So it works like a charm.
Please note that if the user has unsaved changes in Adrenalin, the changes made from ADLX are not taken into account until the user saves the settings in Adrenalin (which are overwritten as soon as ADLX takes over anyway).
We should see if it is possible to override the fact that the user has not saved his configuration and all the cases of figures would indeed allow to modify the speed of rotation of the fans.
@Rem0o

Rem0o, 2023-02-01:
" Set target fan speed is supported " always return false whatever state you were in the first place?

I wish I could set off the advanced mode and set the fan speed manually. If you can find an initial state where it returns true, would be great.

I didn't find any specific binding equivalent to the "advanced mode" toggle.  Otherwise, I guess we can make it work by changing the state list points every time....

Rem0o, 2023-02-01:
Added a test loop at the end which will try and set speeds from 20% to 100%.

antt0n, 2023-02-01:
> Added a test loop at the end which will try and set speeds from 20% to 100%.

All the time this error : `ADLX result (ADLX_NOT_SUPPORTED): Could not get target Fan speed`
I will upload a youtube video where I test all the possibilities of the previous commit

antt0n, 2023-02-01:
https://youtu.be/Jm3PdwtBr8A
@Rem0o Here is a video that allows you to see the output with the previous commit of several different cases on Adrenalin.

Rem0o, 2023-02-01:
@antt0n 

> All the time this error : ADLX result (ADLX_NOT_SUPPORTED): Could not get target Fan speed

Cleared that return condition.

Can you test the set fan speed loop of my latest commit?

antt0n, 2023-02-01:
@Rem0o 

> @antt0n
> 
> > All the time this error : ADLX result (ADLX_NOT_SUPPORTED): Could not get target Fan speed
> 
> Cleared that return condition.
> 
> Can you test the set fan speed loop of my latest commit?

[image]
[image]

Rem0o, 2023-02-01:
Tried to skip a few steps. Just committed again with a more straight forward approach.

antt0n, 2023-02-02:
> Tried to skip a few steps. Just committed again with a more straight forward approach.

[image]
[image]
[image]

Same as yesterday

Rem0o, 2023-02-02:
@antt0n  is the program actively changing the dots in the graph to all 50%?  Just committed again.

antt0n, 2023-02-02:
> @antt0n is the program actively changing the dots in the graph to all 50%? Just committed again.

The points of the graph were updated only once at 50% with the previous commit.
With the new commit you sent, it works by gradually increasing the speed of the fans.
```
Set target fan speed is supported: False
ADLX result (ADLX_NOT_SUPPORTED): Could not get target Fan speed
Current zero RPM status: True
Zero RPM supported: True
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
States error index is -1
Setting fan speed: 20
States error index is -1
Setting fan speed: 21
States error index is -1
Setting fan speed: 22
States error index is -1
Setting fan speed: 23
States error index is -1
Setting fan speed: 24
States error index is -1
Setting fan speed: 25
States error index is -1
Setting fan speed: 26
States error index is -1
Setting fan speed: 27
States error index is -1
Setting fan speed: 28
States error index is -1
Setting fan speed: 29
States error index is -1
Setting fan speed: 30
States error index is -1
Setting fan speed: 31
States error index is -1
Setting fan speed: 32
States error index is -1
Setting fan speed: 33
States error index is -1
Setting fan speed: 34
States error index is -1
Setting fan speed: 35
States error index is -1
Setting fan speed: 36
States error index is -1
Setting fan speed: 37
States error index is -1
Setting fan speed: 38
States error index is -1
Setting fan speed: 39
States error index is -1
Setting fan speed: 40
States error index is -1
Setting fan speed: 41
States error index is -1
Setting fan speed: 42
States error index is -1
Setting fan speed: 43
States error index is -1
Setting fan speed: 44
States error index is -1
Setting fan speed: 45
States error index is -1
Setting fan speed: 46
States error index is -1
Setting fan speed: 47
States error index is -1
Setting fan speed: 48
States error index is -1
Setting fan speed: 49
States error index is -1
Setting fan speed: 50
States error index is -1
Setting fan speed: 51
States error index is -1
Setting fan speed: 52
States error index is -1
Setting fan speed: 53
States error index is -1
Setting fan speed: 54
States error index is -1
Setting fan speed: 55
States error index is -1
Setting fan speed: 56
States error index is -1
Setting fan speed: 57
States error index is -1
Setting fan speed: 58
States error index is -1
Setting fan speed: 59
States error index is -1
Setting fan speed: 60
States error index is -1
Setting fan speed: 61
States error index is -1
Setting fan speed: 62
States error index is -1
Setting fan speed: 63
States error index is -1
Setting fan speed: 64
States error index is -1
Setting fan speed: 65
States error index is -1
Setting fan speed: 66
States error index is -1
Setting fan speed: 67
States error index is -1
Setting fan speed: 68
States error index is -1
Setting fan speed: 69
States error index is -1
Setting fan speed: 70
States error index is -1
Setting fan speed: 71
States error index is -1
Setting fan speed: 72
States error index is -1
Setting fan speed: 73
States error index is -1
Setting fan speed: 74
States error index is -1
Setting fan speed: 75
States error index is -1
Setting fan speed: 76
States error index is -1
Setting fan speed: 77
States error index is -1
Setting fan speed: 78
States error index is -1
Setting fan speed: 79
States error index is -1
Setting fan speed: 80
States error index is -1
Setting fan speed: 81
States error index is -1
Setting fan speed: 82
States error index is -1
Setting fan speed: 83
States error index is -1
Setting fan speed: 84
States error index is -1
Setting fan speed: 85
States error index is -1
Setting fan speed: 86
States error index is -1
Setting fan speed: 87
States error index is -1
Setting fan speed: 88
States error index is -1
Setting fan speed: 89
States error index is -1
Setting fan speed: 90
States error index is -1
Setting fan speed: 91
States error index is -1
Setting fan speed: 92
States error index is -1
Setting fan speed: 93
States error index is -1
Setting fan speed: 94
States error index is -1
Setting fan speed: 95
States error index is -1
Setting fan speed: 96
States error index is -1
Setting fan speed: 97
States error index is -1
Setting fan speed: 98
States error index is -1
Setting fan speed: 99
Temp is 36
Hotspot Temp is 39
Fan speed is 3142
```

antt0n, 2023-02-02:
@Rem0o 
https://youtu.be/Y2WSrPwtUDc

Rem0o, 2023-02-03:
Awesome. Figured out there is a call to get the legal fan speed range. Applied that to the program.

xhemp, 2023-02-03:
Set target fan speed is supported: False
ADLX result (ADLX_NOT_SUPPORTED): Could not get target Fan speed
Current zero RPM status: False
Zero RPM supported: True
States error index is -1
Setting fan speed: 20
States error index is -1
Setting fan speed: 21
States error index is -1
Setting fan speed: 22
States error index is -1
Setting fan speed: 23
States error index is -1
Setting fan speed: 24
States error index is -1
Setting fan speed: 25
States error index is -1
Setting fan speed: 26
States error index is -1
Setting fan speed: 27
States error index is -1
Setting fan speed: 28
States error index is -1
Setting fan speed: 29
States error index is -1
..
Setting fan speed: 96
States error index is -1
Setting fan speed: 97
States error index is -1
Setting fan speed: 98
States error index is -1
Setting fan speed: 99
Temp is 30
Hotspot Temp is 36
Fan speed is 3393
It goes all the way up to 99%. (along with the fan).

antt0n, 2023-02-03:
> Awesome. Figured out there is a call to get the legal fan speed range. Applied that to the program.
> 
> There is much optimization and stuff to clean up and be done, but for the most part, I think we got a proof of concept that fan control and monitoring can be done with this new API.

The last commit works relatively well.
When testing a little further, it seems that when using the preconfigured settings, the program fails to automatically switch to manual settings.
[image]

Rem0o, 2023-02-03:
@antt0n  so you have to be in manual mode in the first place?

Rem0o, 2023-02-05:
Wrapped the c#/C++ bindings in classes so it's easier to play with.  Not exactly done yet but I got the fan stuff done. Redid the main program with the wrapper.   You can play with the code more easily if you want to.

antt0n, 2023-02-07:
> @antt0n so you have to be in manual mode in the first place? I'm looking through the test program in C/C++, and they don't do any particular to initiate manual control.

Sorry for the delay in responding. That's not quite right.
I need to be out of "Silent" (silenieux) or "balanced" (équilibré) mode. Everyone else switches to manual mode fine, but not them.

Rem0o, 2023-02-07:
No worries.  I'm not sure if you are aware, but I develop FanControl based on LHM. Made a plugin out of ADLX.
https://ci.appveyor.com/project/Rem0o/fancontrol-adlx/build/artifacts

If you could give it a shot, would be appreciated. You'll probably need to right-click property => unlock all the dlls. 

Also, out of the test project you tested a while ago (https://gpuopen.com/manuals/ADLX/adlx-cpp__manual_fan_tuning/#code), was there something in there that allowed to switched between the different presets (balanced, silent etc) ?

antt0n, 2023-02-07:
> No worries. I'm not sure if you are aware, but I develop FanControl based on LHM. Made a plugin out of ADLX. https://ci.appveyor.com/project/Rem0o/fancontrol-adlx/build/artifacts
>
> If you could give [...]

[image]

In Plugins folder (after unlock dlls)
[image]

Rem0o, 2023-02-07:
Dumb mistake, my bad. Updated the plugin and wrapper.

antt0n, 2023-02-07:
> Dumb mistake, my bad. Updated the plugin and wrapper.

The software can no longer start with the latest version.
I obviously have no logs.

Rem0o, 2023-02-07:
That's odd, is the process running at all? Wipe the plugin folder clean you should get back on your feet.

antt0n, 2023-02-07:
> That's odd, is the process running at all? Wipe the plugin folder clean you should get back on your feet.

Yes, that's what I did.
By deleting the contents of the plugin folder everything starts correctly

Rem0o, 2023-02-07:
Try again and maybe take a look at the log or event viewer? If it ain't starting at all it should have some logs in either of these places.

xhemp, 2023-02-10:
This is have in the logs when I run FanControl with the 3 DLLs in place:

10/02/2023 12:15:42: System.Runtime.InteropServices.SEHException (0x80004005): External component has thrown an exception.
   at ADLXWrapper.Bindings.ADLXPINVOKE.IADLXGPUTuningServices_GetManualFanTuning(HandleRef jarg1, HandleRef jarg2, HandleRef jarg3)
   at ADLXWrapper.Bindings.IADLXGPUTuningServices.GetManualFanTuning(IADLXGPU pGPU, SWIGTYPE_p_p_adlx__IADLXInterface ppManualFanTuning)
   at ADLXWrapper.GPUTuningService.GetManualFanTuning(GPU gpu)
   at System.Linq.Enumerable.WhereSelectListIterator`2.MoveNext()
   at System.Linq.Buffer`1..ctor(IEnumerable`1 source)
   at System.Linq.Enumerable.ToArray[TSource](IEnumerable`1 source)
   at FanControl.ADLX.ADLXPlugin.Initialize()

Rem0o, 2023-02-10:
@xhemp 

2 things: 
1) Make sure all the files in the plugin folder you copy are unblocked. (Right-click, properties)
2) I added an extra step to check if "GetManualFanTuning" is supported for your gpu in the first place, with logs.

antt0n, 2023-02-10:
> 

```
AVT : la journalisation de liaison d'assembly est désactivée.
Pour activer la journalisation des échecs de liaison d'assembly, attribuez la valeur 1 à la valeur de Registre [HKLM\Software\Microsoft\Fusion!EnableLog] (DWORD).
Remarque : une certaine perte de performance est associée à l'enregistrement dans le journal des échecs de liaison d'assembly.
Pour désactiver cette fonctionnalité, supprimez la valeur de Registre [HKLM\Software\Microsoft\Fusion!EnableLog].

07/02/2023 17:23:41: System.IO.FileNotFoundException: Impossible de charger le fichier ou l'assembly 'ADLXWrapper, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null' ou une de ses dépendances. Le fichier spécifié est introuvable.
Nom de fichier : 'ADLXWrapper, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null'
   à FanControl.ADLX.ADLXPlugin.Close()
   à System.Collections.Generic.List`1.ForEach(Action`1 action)
   à FanControl.Domain.ComputerAccessLayer.Close()
   à FanControl.Domain.ComputerAccessLayer.<>c__DisplayClass11_0.<<TryRefresh>b__0>d.MoveNext()

AVT : la journalisation de liaison d'assembly est désactivée.
Pour activer la journalisation des échecs de liaison d'assembly, attribuez la valeur 1 à la valeur de Registre [HKLM\Software\Microsoft\Fusion!EnableLog] (DWORD).
Remarque : une certaine perte de performance est associée à l'enregistrement dans le journal des échecs de liaison d'assembly.
Pour désactiver cette fonctionnalité, supprimez la valeur de Registre [HKLM\Software\Microsoft\Fusion!EnableLog].

07/02/2023 17:23:41: Initialization failed.
07/02/2023 17:25:44: Unhandled exception in FanControl v1.0.0.0
07/02/2023 17:25:44: System.IO.FileNotFoundException: Impossible de charger le fichier ou l'assembly 'ADLXWrapper, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null' ou une de ses dépendances. Le fichier spécifié est introuvable.
Nom de fichier : 'ADLXWrapper, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null'
   à FanControl.ADLX.ADLXPlugin.Close()
   à System.Collections.Generic.List`1.ForEach(Action`1 action)
   à FanControl.Domain.ComputerAccessLayer.Close()
   à Autofac.Core.Disposer.Dispose(Boolean disposing)
   à Autofac.Util.Disposable.Dispose()
   à Autofac.Core.Lifetime.LifetimeScope.Dispose(Boolean disposing)
   à Autofac.Util.Disposable.Dispose()
   à Autofac.Core.Container.Dispose(Boolean disposing)
   à Autofac.Util.Disposable.Dispose()
   à FanControl.App.OnExit(ExitEventArgs e)
   à System.Windows.Application.DoShutdown()
   à System.Windows.Application.ShutdownImpl()
   à System.Windows.Application.ShutdownCallback(Object arg)
   à System.Windows.Threading.ExceptionWrapper.InternalRealCall(Delegate callback, Object args, Int32 numArgs)
   à System.Windows.Threading.ExceptionWrapper.TryCatchWhen(Object source, Delegate callback, Object args, Int32 numArgs, Delegate catchHandler)

AVT : la journalisation de liaison d'assembly est désactivée.
Pour activer la journalisation des échecs de liaison d'assembly, attribuez la valeur 1 à la valeur de Registre [HKLM\Software\Microsoft\Fusion!EnableLog] (DWORD).
Remarque : une certaine perte de performance est associée à l'enregistrement dans le journal des échecs de liaison d'assembly.
Pour désactiver cette fonctionnalité, supprimez la valeur de Registre [HKLM\Software\Microsoft\Fusion!EnableLog].

07/02/2023 17:25:48: System.NullReferenceException: La référence d'objet n'est pas définie à une instance d'un objet.
   à ADLXWrapper.GPU..ctor(IADLXGPU gpu)
   à ADLXWrapper.SystemServices.GetGPUs()
   à FanControl.ADLX.ADLXPlugin.Initialize()
07/02/2023 17:25:52: System.NullReferenceException: La référence d'objet n'est pas définie à une instance d'un objet.
   à ADLXWrapper.GPU..ctor(IADLXGPU gpu)
   à ADLXWrapper.SystemServices.GetGPUs()
   à FanControl.ADLX.ADLXPlugin.Initialize()
07/02/2023 17:25:55: ADLX could not initialize or has no sensors.
07/02/2023 17:26:21: System.NullReferenceException: La référence d'objet n'est pas définie à une instance d'un objet.
   à ADLXWrapper.GPU..ctor(IADLXGPU gpu)
   à ADLXWrapper.SystemServices.GetGPUs()
   à FanControl.ADLX.ADLXPlugin.Initialize()
07/02/2023 17:26:24: System.NullReferenceException: La référence d'objet n'est pas définie à une instance d'un objet.
   à ADLXWrapper.GPU..ctor(IADLXGPU gpu)
   à ADLXWrapper.SystemServices.GetGPUs()
   à FanControl.ADLX.ADLXPlugin.Initialize()
07/02/2023 17:26:27: ADLX could not initialize or has no sensors.
07/02/2023 17:27:23: System.NullReferenceException: La référence d'objet n'est pas définie à une instance d'un objet.
   à ADLXWrapper.GPU..ctor(IADLXGPU gpu)
   à ADLXWrapper.SystemServices.GetGPUs()
   à FanControl.ADLX.ADLXPlugin.Initialize()
07/02/2023 17:27:59: System.NullReferenceException: La référence d'objet n'est pas définie à une instance d'un objet.
   à ADLXWrapper.GPU..ctor(IADLXGPU gpu)
   à ADLXWrapper.SystemServices.GetGPUs()
   à FanControl.ADLX.ADLXPlugin.Initialize()
07/02/2023 17:28:02: System.NullReferenceException: La référence d'objet n'est pas définie à une instance d'un objet.
   à ADLXWrapper.GPU..ctor(IADLXGPU gpu)
   à ADLXWrapper.SystemServices.GetGPUs()
   à FanControl.ADLX.ADLXPlugin.Initialize()
07/02/2023 17:28:05: ADLX could not initialize or has no sensors.
07/02/2023 17:29:29: System.NullReferenceException: La référence d'objet n'est pas définie à une instance d'un objet.
   à ADLXWrapper.GPU..ctor(IADLXGPU gpu)
   à ADLXWrapper.SystemServices.GetGPUs()
   à FanControl.ADLX.ADLXPlugin.Initialize()
[... cut ...]
07/02/2023 17:36:51: System.NullReferenceException: La référence d'objet n'est pas définie à une instance d'un objet.
   à ADLXWrapper.GPU..ctor(IADLXGPU gpu)
   à ADLXWrapper.SystemServices.GetGPUs()
   à FanControl.ADLX.ADLXPlugin.Initialize()
07/02/2023 17:36:54: System.NullReferenceException: La référence d'objet n'est pas définie à une instance d'un objet.
   à ADLXWrapper.GPU..ctor(IADLXGPU gpu)
   à ADLXWrapper.SystemServices.GetGPUs()
   à FanControl.ADLX.ADLXPlugin.Initialize()
07/02/2023 17:36:58: ADLX could not initialize or has no sensors.
```

Rem0o, 2023-02-10:
@antt0n  are you sure ADLXWrapper is in the plugin folder? Logs says it's missing somehow.

antt0n, 2023-02-10:
> @antt0n are you sure ADLXWrapper is in the plugin folder? Logs says it's missing somehow.

[image]

Everything is correctly installed.
At first the application started but did not detect the plugin, now the application does not start and is not launched in the background.

xhemp, 2023-02-10:
> @xhemp
> 
> 2 things:
> 
>     1. Make sure all the files in the plugin folder you copy are unblocked. (Right-click, properties)
> 
>     2. I added an extra step to check if "GetManualFanTuning" is supported for your gpu in the first place, with logs.

I have unblocked the DLLs.
This is what I got now:

```
10/02/2023 15:33:17: System.Runtime.InteropServices.SEHException (0x80004005): External component has thrown an exception.
   at ADLXWrapper.Bindings.ADLXPINVOKE.IADLXGPUTuningServices_IsSupportedManualFanTuning(HandleRef jarg1, HandleRef jarg2, HandleRef jarg3)
   at ADLXWrapper.Bindings.IADLXGPUTuningServices.IsSupportedManualFanTuning(IADLXGPU pGPU, SWIGTYPE_p_bool supported)
   at ADLXWrapper.GPUTuningService.IsManualFanTuningSupported(GPU gpu)
   at FanControl.ADLX.ADLXPlugin.<Initialize>b__15_0(GPU g)
   at System.Linq.Enumerable.WhereSelectListIterator`2.MoveNext()
   at System.Linq.Buffer`1..ctor(IEnumerable`1 source)
   at System.Linq.Enumerable.ToArray[TSource](IEnumerable`1 source)
   at FanControl.ADLX.ADLXPlugin.Initialize()
```

Rem0o, 2023-02-10:
@antt0n 

make sure you got all dlls in the plugin folder unblocked
[image]

Rem0o, 2023-02-10:
@xhemp  if you take a step back and run the test program instead of the plugin, what do you got.

antt0n, 2023-02-10:
> @antt0n
> 
> make sure you got all dlls in the plugin folder unblocked [image]

I'm sure the DLLs are unlocked, and the folder is not in drive alone.

Rem0o, 2023-02-10:
@antt0n  same thing for you,
 
 Maybe take a step back and run the test console again, (update the source),  here https://github.com/Rem0o/ADLX/tree/main/Samples/csharp

xhemp, 2023-02-10:
> @xhemp if you take a step back and run the test program instead of the plugin, what do you got.

Do you mean the ADLXWrapper? Here's the log:

```
Set target fan speed is supported: False
ADLX result (ADLX_NOT_SUPPORTED): Could not get target Fan speed
Current zero RPM status: True
Zero RPM supported: True
States error index is -1
Setting fan speed: 20
States error index is -1
Setting fan speed: 21
States error index is -1
Setting fan speed: 22
States error index is -1
Setting fan speed: 23
States error index is -1
Setting fan speed: 24
States error index is -1
Setting fan speed: 25
States error index is -1
Setting fan speed: 26
States error index is -1
Setting fan speed: 27
States error index is -1
Setting fan speed: 28
States error index is -1
Setting fan speed: 29
States error index is -1
Setting fan speed: 30
States error index is -1
Setting fan speed: 31
States error index is -1
Setting fan speed: 32
States error index is -1
Setting fan speed: 33
States error index is -1
Setting fan speed: 34
States error index is -1
Setting fan speed: 35
States error index is -1
Setting fan speed: 36
...
States error index is -1
Setting fan speed: 96
States error index is -1
Setting fan speed: 97
States error index is -1
Setting fan speed: 98
States error index is -1
Setting fan speed: 99
Temp is 31
Hotspot Temp is 38
Fan speed is 3388

```

Rem0o, 2023-02-10:
@xhemp  did you update the source?  Made a loop with the ADLXWrapper, and with the bindings directly.  

I suspect one of the problem for the plugin is the dll locations relative to fancontrol itself.  Try and copy the ADLXBinding dll stuff onto the fancontrol folder directly, and keep the FanControl.ADLX.dll + the ADLXWrapper.dll in the plugin folder.

xhemp, 2023-02-10:
> @xhemp did you update the source? Made a loop with the ADLXWrapper, and with the bindings directly.
>
> I suspect one of the problem for the plugin is the dll locations relative to fancontrol itself. [...]

Yes, I just refetched, ran again and this what I get:
```

Set target fan speed is supported: False
ADLX result (ADLX_NOT_SUPPORTED): Could not get target Fan speed
Current zero RPM status: False
Zero RPM supported: True
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
ADLX result (ADLX_INVALID_ARGS): Couldn't validate states
States error index is 0
```

With regards to FanControl, I copied ADLXCSharpBind.dll to the FanControl main folder and tried to run but the log is the same:

```
10/02/2023 15:51:37: System.Runtime.InteropServices.SEHException (0x80004005): External component has thrown an exception.
   at ADLXWrapper.Bindings.ADLXPINVOKE.IADLXGPUTuningServices_IsSupportedManualFanTuning(HandleRef jarg1, HandleRef jarg2, HandleRef jarg3)
   at ADLXWrapper.Bindings.IADLXGPUTuningServices.IsSupportedManualFanTuning(IADLXGPU pGPU, SWIGTYPE_p_bool supported)
   at ADLXWrapper.GPUTuningService.IsManualFanTuningSupported(GPU gpu)
   at FanControl.ADLX.ADLXPlugin.<Initialize>b__15_0(GPU g)
   at System.Linq.Enumerable.WhereSelectListIterator`2.MoveNext()
   at System.Linq.Buffer`1..ctor(IEnumerable`1 source)
   at System.Linq.Enumerable.ToArray[TSource](IEnumerable`1 source)
   at FanControl.ADLX.ADLXPlugin.Initialize()
```

xhemp, 2023-02-10:
Saw a couple of other changes on github, pulled and ran it again:

```
Testing with direct bindinges...
Set target fan speed is supported: False
ADLX result (ADLX_NOT_SUPPORTED): Could not get target Fan speed
Current zero RPM status: True
Zero RPM supported: True
Trying to set fan speed to 20
States error index is -1
Setting fan speed: 20
Trying to set fan speed to 21
States error index is -1
Setting fan speed: 21
Trying to set fan speed to 22
States error index is -1
Setting fan speed: 22
Trying to set fan speed to 23
States error index is -1
Setting fan speed: 23
Trying to set fan speed to 24
States error index is -1
Setting fan speed: 24
Trying to set fan speed to 25
States error index is -1
Setting fan speed: 25
Trying to set fan speed to 26
States error index is -1
Setting fan speed: 26
Trying to set fan speed to 27
States error index is -1
Setting fan speed: 27
...
Setting fan speed: 91
Trying to set fan speed to 92
States error index is -1
Setting fan speed: 92
Trying to set fan speed to 93
States error index is -1
Setting fan speed: 93
Trying to set fan speed to 94
States error index is -1
Setting fan speed: 94
Trying to set fan speed to 95
States error index is -1
Setting fan speed: 95
Trying to set fan speed to 96
States error index is -1
Setting fan speed: 96
Trying to set fan speed to 97
States error index is -1
Setting fan speed: 97
Trying to set fan speed to 98
States error index is -1
Setting fan speed: 98
Trying to set fan speed to 99
States error index is -1
Setting fan speed: 99
Temp is 33
Hotspot Temp is 40
Fan speed is 3408
Testing with ADLXWrapper...
Found GPU with name: AMD Radeon RX 7900 XTX

Unhandled Exception: System.Runtime.InteropServices.SEHException: External component has thrown an exception.
   at ADLXWrapper.Bindings.ADLXPINVOKE.IADLXGPUTuningServices_GetManualFanTuning(HandleRef jarg1, HandleRef jarg2, HandleRef jarg3)
   at ADLXWrapper.Bindings.IADLXGPUTuningServices.GetManualFanTuning(IADLXGPU pGPU, SWIGTYPE_p_p_adlx__IADLXInterface ppManualFanTuning) in C:\Users\xhemp\source\repos\ADLX\Samples\csharp\ADLXWrapper\Bindings\IADLXGPUTuningServices.cs:line 122
   at ADLXWrapper.GPUTuningService.GetManualFanTuning(GPU gpu) in C:\Users\xhemp\source\repos\ADLX\Samples\csharp\ADLXWrapper\GPUTuningService.cs:line 23
   at ADLXWrapper.TestConsole.Program.<FanSpeedLoopWithWraper>d__1.MoveNext() in C:\Users\xhemp\source\repos\ADLX\Samples\csharp\ADLXWrapper.TestConsole\Program.cs:line 27
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at ADLXWrapper.TestConsole.Program.<Main>d__0.MoveNext() in C:\Users\xhemp\source\repos\ADLX\Samples\csharp\ADLXWrapper.TestConsole\Program.cs:line 16
--- End of stack trace from previous location where exception was thrown ---
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw()
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification(Task task)
   at ADLXWrapper.TestConsole.Program.<Main>(String[] args)

C:\Users\xhemp\source\repos\ADLX\Samples\csharp\ADLXWrapper.TestConsole\bin\Release\net48\ADLXWrapper.TestConsole.exe (process 161004) exited with code -532462766.
```

Rem0o, 2023-02-10:
Great @xhemp , means I'm doing something wrong with the wrapper. Will investigate...

Rem0o, 2023-02-10:
Saw a difference where I disposed of the unmanaged pointer too early ( I think ) relative to what I was doing with the direct bindings. Commited.

xhemp, 2023-02-10:
Great, here it is:

```
Testing with direct bindinges...
Set target fan speed is supported: False
ADLX result (ADLX_NOT_SUPPORTED): Could not get target Fan speed
Current zero RPM status: False
Zero RPM supported: True
Trying to set fan speed to 20
States error index is -1
Setting fan speed: 20
Trying to set fan speed to 21
States error index is -1
Setting fan speed: 21
Trying to set fan speed to 22
States error index is -1
Setting fan speed: 22
Trying to set fan speed to 23
States error index is -1
Setting fan speed: 23
Trying to set fan speed to 24
States error index is -1
Setting fan speed: 24
Trying to set fan speed to 25
States error index is -1
Setting fan speed: 25
Trying to set fan speed to 26
States error index is -1
Setting fan speed: 26
Trying to set fan speed to 27
States error index is -1
Setting fan speed: 27
Trying to set fan speed to 28
States error index is -1
Setting fan speed: 28
Trying to set fan speed to 29
States error index is -1
Setting fan speed: 29
Trying to set fan speed to 30
States error index is -1
Setting fan speed: 30
Trying to set fan speed to 31
States error index is -1
Setting fan speed: 31
Trying to set fan speed to 32
States error index is -1
Setting fan speed: 32
Trying to set fan speed to 33
States error index is -1
Setting fan speed: 33
Trying to set fan speed to 34
States error index is -1
Setting fan speed: 34
Trying to set fan speed to 35
States error index is -1
Setting fan speed: 35
Trying to set fan speed to 36
States error index is -1
Setting fan speed: 36
Trying to set fan speed to 37
States error index is -1
Setting fan speed: 37
Trying to set fan speed to 38
States error index is -1
Setting fan speed: 38
Trying to set fan speed to 39
States error index is -1
Setting fan speed: 39
Trying to set fan speed to 40
States error index is -1
[... 242 lines cut ...]
Setting fan speed 79%
Setting fan speed 80%
Setting fan speed 81%
Setting fan speed 82%
Setting fan speed 83%
Setting fan speed 84%
Setting fan speed 85%
Setting fan speed 86%
Setting fan speed 87%
Setting fan speed 88%
Setting fan speed 89%
Setting fan speed 90%
Setting fan speed 91%
Setting fan speed 92%
Setting fan speed 93%
Setting fan speed 94%
Setting fan speed 95%
Setting fan speed 96%
Setting fan speed 97%
Setting fan speed 98%
Setting fan speed 99%
Setting fan speed 100%

C:\Users\xhemp\source\repos\ADLX\Samples\csharp\ADLXWrapper.TestConsole\bin\Release\net48\ADLXWrapper.TestConsole.exe (process 161880) exited with code 0.
```

Rem0o, 2023-02-10:
@xhemp  and I would guess the fan speed did indeed increase as the program said?

xhemp, 2023-02-10:
> @xhemp  and I would guess the fan speed did indeed increase as the program said?
> 
> Updated the fan control plugin.

It did, yes, on both instances.

Rem0o, 2023-02-10:
> It did, yes, on both instances.

Good, test out the ADLX Fancontrol plugin if you got the time.

xhemp, 2023-02-10:
> > It did, yes, on both instances.
> 
> Good, test out the ADLX Fancontrol plugin if you got the time.

Tried with the ADLXWrapper dll inside the fancontrol directory and inside the plugin directory, the error is the same:

```
10/02/2023 22:54:53: System.NullReferenceException: Object reference not set to an instance of an object.
   at FanControl.ADLX.ADLXTemperatureSensor.Update()
   at FanControl.ADLX.ADLXTemperatureSensor..ctor(String name, GPU gpu, Func`1 getTemp)
   at FanControl.ADLX.ADLXPlugin.<>c.<Load>b__16_2(GPU gpu, GPUMetrics m)
   at System.Linq.Enumerable.<ZipIterator>d__61`3.MoveNext()
   at System.Linq.Buffer`1..ctor(IEnumerable`1 source)
   at System.Linq.Enumerable.ToArray[TSource](IEnumerable`1 source)
   at FanControl.ADLX.ADLXPlugin.Load(IPluginSensorsContainer _container)
   at FanControl.Domain.BackendProviders.Plugin.PluginBackendProvider.Open()
```

[11 more replies]