LibreHardwareMonitor/LibreHardwareMonitor #836: Latest Win 11 breaks taskbar placement

ez009, 2022-10-05:
Hi. thanks so much for this mod. 
Just did a Win11 update this week and now the taskbar indicator overlaps the last two taskbar icons, covering them up.
If i drag two unused taskbar items to the far right, i am able to see the currently active taskbar icon., and the upload/download indicator overlaps those (which will work as a temp fix) but if you could figure out what they changed to make this act as expected, it'd be much appreciated. 
WinVer: 22H2 (OS Build 22621.608)

THX!

ez009, 2022-10-06:
THX for the quick fix !!

kontul-dev, 2022-10-13:
Hi, I have (probably) related/similar problem with taskbar. I have 3 sensor graphs shown in tray, in particular order, and after I sleep/hibernate/restart my laptop, 2 of them are hidden and all are in changed order.

kontul-dev, 2022-10-26:
Related issue id described here: https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/848