LibreHardwareMonitor/LibreHardwareMonitor #843: not showing details of x670 motherboards

RabbitChowder, 2022-10-09:
[image]
My mobo is runnning relatively new bios. Do I need the newest bios, or is an libre update required? Thanks so much for any help

GlitterKill, 2022-10-14:
Same here.  Nothing detected on X670E motherboard.
[image]

Alexx999, 2022-10-16:
For the X670E Hero it's not picking up SuperIO with the following:

Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0xD802 at 0x2E/0x2F
Chip ID: Unknown Winbond / Nuvoton / Fintek with ID 0x8883 at 0x4E/0x4F
Chip ID: Unknown ITE with ID 0x8883 at 0x4E/0x4F
Chip ID: Unknown SMSC with ID 0x8883 at 0x4E/0x4F

But I'm not sure what's next

jstar81, 2022-10-19:
Is this issue a reason why I am unable to see any temperatures in FanControl software https://github.com/Rem0o/ as it uses by default Libre Hardware Monitor plugin?

Alexx999, 2022-10-19:
@jstar81 yes, the PR above is required for FanControl to work with X670E Hero. However, for me, it doesn't read all temps anyway (or reads them incorrectly).

Inverness, 2022-10-21:
I'm also experiencing this myself:
[image]

I hope to see that PR merged soon.

NerdMayhem, 2022-10-26:
Same for ASUS TUF GAMING X670E-PLUS WIFI
[image]

nirurin, 2022-10-26:
Same for Asus Strix X670e-e. 

Any news on if this is being worked on? Seems to be for all Asus boards only, as my MSI MPG X670 worked fine.

andino66, 2023-01-31:
I'm having the same issues. I have an Asus ROG Crosshair x670E Hero.

jstar81, 2023-01-31:
> I'm having the same issues. I have an Asus ROG Crosshair x670E Hero.

X670E Hero has been supported for a long time, are you using latest version? Please disable (uninstall) possible conflicting / overlapping hardware monitor softwares.