LibreHardwareMonitor/LibreHardwareMonitor #852: fan not detected on gigabyte x670e aorus master

sagy101, 2022-10-22:
hi,
i had my fans connected each to its separate header - 
cpu_fan (3 fans daisy chained)
sys_fan1
sys_fan2
sys_fan3
sys_fan6_pump
sys_fan7_pump
sys_fan8_pump

and hwmonitor only detected cpu_fan, sys_fan1, sys_fan2, sys_fan3 so my first thought was to daisy chain the 3 fans that were not detected to my sys_fan3 since its the same as sys_fan1/2/3 which are detected.

that did not work, so i thought maybe it isnt in pwm mode, so i went to bios and changed it from auto mode to pwm.
that did not work as well.

any suggestions as to what is the problem, and how to solve it? i wanna use fanControl and from what i understand it uses hwMonitor so...

system spec:
Ryzen 7900x
Gigabyte x670e aorus master (f7 bios version)
all fans are the same arctic P12 PWM PST A-RGB fans
i think that's whats relevant.

PStanos, 2023-05-21:
I am having the same issue, and I'm currently on the latest F10 BIOS.

HWiNFO64 is only reporting CPU, CPU_OPT, and System 1 under `ITE IT8689E`, and System 7 Pump is reported under `Gigabyte EC: Gigabyte X670E AUROS MASTER`. No other headers are listed.

I am using Fan Control, and under the LibreHardwareMonitor section I tried enabling more features, including `InpOutput`, but nothing has worked yet. I'm able to control CPU, CPU_OPT, and System 1 fans just fine, but none of the other 4 case fans.

antt0n, 2023-06-07:
Hello,

I have exactly the same problem.

jilherme, 2023-07-03:
I have the same problem with my B650M aorus elite AX. Can only control CPUfan. SYS_FAN 1, 2 and 3 aren't available
[image]

jilherme, 2023-08-26:
> I have the same problem with my B650M aorus elite AX. Can only control CPUfan. SYS_FAN 1, 2 and 3 aren't available [image]

It's now working now! In FanControl too.
[image]

progenitor-amborella, 2024-04-03:
I also have this issue. Gigabyte B650 Aero G, fan config as follows:
```
cpu_fan: PWM fan - shown
cpu_opt: voltage fan - shown
sys_fan1: PWM fan - shown
sys_fan2: voltage fan -shown
sys_fan3: voltage fan - shown
sys_fan4: voltage fan - not shown
sys_fan5_pump: PWM fan - not shown
sys_fan6_pump: unused
```
Specifically the software I am using for this is [FanControl](https://github.com/Rem0o/FanControl.Releases). Utilizing the instructions [here](https://github.com/Rem0o/FanControl.HWInfo) to integrate HWiNFO64 sensors does expose the RPM data to FanControl through the use of a plugin, but as one might expect, does not allow for control.