LibreHardwareMonitor/LibreHardwareMonitor #857: 13th Gen Intel Core i9-13900KF can't get temperature

Quttia, 2022-10-25:
[HardwareReport.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/9856297/HardwareReport.txt)

MGD4096, 2022-11-11:
run app as administrator

Aytackydln, 2022-11-11:
It is definitely not about administrator privilage. Project-Aurora runs as admin and users with same cpu have this problem, too.

MGD4096, 2022-11-11:
In my case, I needed administrator privileges to see the CPU temperatures.

[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/9992880/LibreHardwareMonitor.Report.txt)

Quttia, 2022-11-12:
The LibreHardwareMonitorLib.dll version I used in this issue was 0.9.0, now I upgrade to 0.9.1, problem solved. Thanks all.

closed 2022-11-12 by Quttia