LibreHardwareMonitor/LibreHardwareMonitor #888: Cpu Total Usage not match with task manager

KrishnagaanthAlphaGhost, 2022-11-14:
Hi Developer,

In Cpu->Load->Cpu Total is differ from Taskmanager cpu use how to match with it for my application,
help me for sort out it

farkeld, 2022-11-25:
I have the same issue. I ran a stress test and even when all cores are at 100%, LHM shows percentages way below and most of them seem totally unrelated with the cpu behavior

MathieuVillegas, 2022-11-26:
It's most probably Windows 11 task manager which displays a CPU usage which  which is not the real value. For example, if you play a game on a 12 cores CPU and the came use only one core at 100%, Windows task manager will report a CPU usage of 90% or so, while in fact less than 10% of your raw CPU power is used, and Libre hardware monitor will report less than 10%. In fact it's very easy to see how the task manager information is misleading. Just open the perfmon  (Performance Monitor) software to see CPU usage, and you will see that the values will be completely different from Task Manager and will match Libre Hardware monitor ones.

closed 2022-12-07 by KrishnagaanthAlphaGhost