LibreHardwareMonitor/LibreHardwareMonitor #893: MSI A520M-A PRO w/ AMD Ryzen 5 5600G: "Powers" issues: using wrong unit, very volatile readings, some make no sense

ralfbergs, 2022-11-15:
Libre Hardware Monitor version 0.9.1

My hardware:

- MoBo MSI A520M-A PRO
- BIOS: 1.B0 (latest as of today)
- CPU AMD Ryzen 5 5600G with Radeon Graphics
- RAM: 32 GB, 2 x 16 GB DDR4-3200, A-DATA Technology

I'm seeing the following issues on my setup:

- The "Powers" values obviously use the wrong units. LHM shows "Powers" of up to "65xxx W" for the CPU "Package." It is pretty obvious that these are milliwatts, my CPU has a TDP of 65 W.
- The readings for the CPU "Powers" are also extremely volatile and drop to "0 W". I absolutely realize that the CPU clock is changed dynamically depending on compute requirements, but can the power really drop to zero, while the PC is powered up? I really doubt this.
Screenshot:
[image]
- The "Power" readings for the different cores also seem to make little sense. You see max values of more than 100 W for a single core, which in my understanding cannot be possible for a TDP of 65 W for the whole CPU. Also, if you add up the wattages shown in the screenshot for all cores you reach more than 260 W, which also seems unreasonable.

Let me know if/how I can help chasing these issues down.

Many thanks for keeping the original project alive with this new fork.

ntoff, 2022-11-19:
Seems like it's probably a duplicate of https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/issues/566

ralfbergs, 2022-11-20:
I agree -- even more so as I also have "memory integrity ENABLED".

So I'm closing this issue. Thanks for your comment.

closed 2022-11-20 by ralfbergs