LibreHardwareMonitor/LibreHardwareMonitor #899: CPU temps/clocks not populated, Installing driver LibreHardwareMonitor.sys failed

craiglandis, 2022-11-20:
Win11 22H2 13900K/ASUS Z790 Hero BIOS 0502 Haven't updated to latest yet (0703).

I'm an admin on the machine, running LibreHardwareMonitor.exe prompts for elevation, I agree, and CPU temps/clocks aren't populated. Everything else appears to be getting populated as expected.

I've got a 12900K/ASUS Z690 Hero and this issue doesn't repro on that - CPU temps and clocks show up fine.

From the attached report:

--------------------------------------------------------------------------------
Ring0

Status: Installing driver "C:\my\LibreHardwareMonitor-net472\LibreHardwareMonitor.sys" failed
First Exception: StartService returned the error: Access is denied. (Exception from HRESULT: 0x80070005 (E_ACCESSDENIED))
Second Exception: StartService returned the error: The system cannot find the file specified. (Exception from HRESULT: 0x80070002)
--------------------------------------------------------------------------------

[LibreHardwareMonitor.Report.txt](https://github.com/LibreHardwareMonitor/LibreHardwareMonitor/files/10048160/LibreHardwareMonitor.Report.txt)

ntoff, 2022-11-22:
>"C:\my\LibreHardwareMonitor-net472\LibreHardwareMonitor.sys"

Do you really have it in a folder called just "my"? Or do you have a directory called like "my xxxxxxxx" with a space in the directory name? Perhaps it doesn't like spaces in the file path?

ntoff, 2022-12-03:
It could be your antivirus blocking it. I've had one pc's windows defender block a file that another pc's windows defender thought was fine. 

It could also be hating the downloaded version because files downloaded from the internet get a special flag that tell windows they're "unsafe".